ARMmbed/mbed-os #13200: mbed-os-6.1 can't compile with IAR

jeromecoutant, 2020-06-29:
### Description of defect

Hi

IAR compilation is failing with latest label:

````
Compile [ 99.8%]: except.S
Compile: C:\Program Files (x86)\IAR Systems\Embedded Workbench 8.2\arm\bin\iasmarm --cpu Cortex-M4F -DSTM32F429xx -D__MBED_CMSIS_RTOS_CM -DARM_MATH_CM4 -DMBED_TEST_MODE -DUSE_FULL_LL_DRIVER -D__CMSIS_RTOS -DUSE_HAL_DRIVER -D__FPU_PRESENT=1 -DTRANSACTION_QUEUE_SIZE_SPI=2 -D__CORTEX_M4 -f .\BUILD\tests\NUCLEO_F429ZI\IAR\asm_macros_fdad6ec39671dad1fefc7fb07a834d55.xcl -o BUILD\tests\NUCLEO_F429ZI\IAR\platform\source\TARGET_CORTEX_M\TOOLCHAIN_IAR\except.o .\platform\source\TARGET_CORTEX_M\TOOLCHAIN_IAR\except.S
[DEBUG] Return: 2
[ERROR]
Failed to build library
````

Issue comes with #12824 

@kjbracey-arm 
@MarceloSalazar 

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

IAR

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.1.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

NA

#### How is this defect reproduced ? 

NA

0xc0170, 2020-06-29:
> [DEBUG] Return: 2

I like the errors like this, just a number 😞  

@kjbracey-arm we should be able to reproduce this just compiling the file, using size limited IAR version?

jeromecoutant, 2020-06-29:
Proposition: https://github.com/jeromecoutant/mbed/commit/c0d12264490fbee7c91a8b0f343e569a929fd54d

0xc0170, 2020-06-30:
>Proposition: jeromecoutant@c0d1226

Can you elaborate , there are 3 things changed, all 3 needed to fix the issue ?

jeromecoutant, 2020-06-30:
> Can you elaborate , there are 3 things changed, all 3 needed to fix the issue ?

I don't really understand...
I just compare commits and add lines that makes compilation OK....

jeromecoutant, 2020-07-07:
> Proposition: [jeromecoutant@c0d1226](https://github.com/jeromecoutant/mbed/commit/c0d12264490fbee7c91a8b0f343e569a929fd54d)

Should I push it in a PR ?
Too late for patch release ?

0xc0170, 2020-07-08:
yes if today, it shall be fine.

I was refering earlier to have better description. I dont understand how adding section, alignment and branching fixes the bug (looks like 3 different issues?).

closed 2020-09-09 by jeromecoutant