ARMmbed/mbed-os #13201: STM32F091 can't pass argument

pilotak, 2020-06-29:
### Description of defect

When running code bellow on F091 there is always following error: functions `pass()` is the problem, exactly the part where i try to read from `data` no matter if it's in `printf` or assigned to a variable.

If this code is run on NUCLEO_F412ZG there is no problem.
```
++ MbedOS Fault Handler ++

FaultType: HardFault

Context:
R0: 200000C5
R1: 20000280
R2: 9
R3: 80001A3
R4: 200000C4
R5: 0
R6: 0
R7: 0
R8: 0
R9: 0
R10: 0
R11: 0
R12: 0
SP   : 20001C28
LR   : 80001A9
PC   : 8000180
xPSR : 1000000
PSP  : 20001C08
MSP  : 20007FB8
CPUID: 410CC200
Mode : Thread
Priv : Privileged
Stack: PSP

-- MbedOS Fault Handler --

++ MbedOS Error Info ++
Error Status: 0x80FF013D Code: 317 Module: 255
Error Message: Fault exception
Location: 0x8000180
Error Value: 0x20000BAC
Current Thread: main Id: 0x200004FC Entry: 0x8001AA9 StackSize: 0x1000 StackMem: 0x20000C50 SP: 0x20001C28
Next:
main  State: 0x2 Entry: 0x8001AA9 Stack Size: 0x1000 Mem: 0x20000C50 SP: 0x20001C10
Ready:
rtx_idle  State: 0x1 Entry: 0x8001BA5 Stack Size: 0x200 Mem: 0x20000990 SP: 0x20000B50
Wait:
rtx_timer  State: 0x83 Entry: 0x8003371 Stack Size: 0x300 Mem: 0x20000690 SP: 0x20000918
Delay:
For more info, visit: https://mbed.com/s/error?error=0x80FF013D&tgt=NUCLEO_F091RC
-- MbedOS Error Info --
```

#### Target(s) affected by this defect ?
NUCLEO_F091RC

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM 

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-99.99.99

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
```cpp
#include "mbed.h"
unsigned char packet[3] = {0, 1, 2};

void pass(const unsigned char *data) {
    printf("pass: %u\n", *(unsigned short *)(data + 1));
}

void test() {
    printf("test: %u\n", *(unsigned short *)(packet + 1));
    pass(packet);
}

int main() {
    test();
    return 0;
}
```

0xc0170, 2020-06-29:
Additional details:

`NUCLEO_F091RC` is Cortex-M0 and `NUCLEO_F412ZG` is Cortex M4F.

Can you test std library (see https://github.com/ARMmbed/mbed-os/blob/master/platform/source/minimal-printf/README.md) to confirm this is an issue with minimal printf implementation?

pilotak, 2020-06-29:
It's not related to `printf` (although i have tried `"target.printf_lib": "std"` as well)

If I alter the code to following, the error is still there
```cpp
void pass(const unsigned char *t) {
    unsigned short var = *(unsigned short *)(t + 1);

    if (var == 1) {
        printf("yes\n");

    } else {
        printf("no\n");
    }
}
```

0xc0170, 2020-06-30:
I see now. 

You are getting a hard fault but due to misaligned access. Cortex M3/M4 tolerate it, but M0 does not. You have 8bit buffer but accessing it via 16bit. Check `The Definitive Guide to the ARM Cortex-M0` , page 83 for details.

I'll close this as resolved.

closed 2020-06-30 by 0xc0170

pilotak, 2020-06-30:
Ok, thank you I did know that