ARMmbed/mbed-os #13203: mbed LPC17xx USB device not working

vvkaggarwal12, 2020-06-29:
### Description of defect
I am working on mbed LPC1768 board.
I build the code for USBSerial Example code for this board.
After flashing the code i am getting below error on UART Serial:

Error␣Status:␣0x80FF0144␣Code:␣324␣Module:␣255
Error␣Message:␣Assertion␣failed:␣status␣==␣osOK
Location:␣0x4295
File:␣.\mbed-os\rtos\source\ThisThread.cpp+225
Error␣Value:␣0x0
Current␣Thread:␣main␣Id:␣0x100014B4␣Entry:␣0x5679␣StackSize:␣0x1000␣StackMem:␣0x10001D98␣SP:␣0x10002CC4␣

And USB is not enumeratin on the host side.
After investigating, in file targets/TARGET_NXP/USBHAL_LPC17xx.cpp, line 414, ThisThread::sleep_for is causing this issue.
When i replaced that line of code with wait_us it works.

#### Target(s) affected by this defect ?
NXP LPC17xx

#### Toolchain(s) (name and version) displaying this defect ?
arm gcc v9

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.1.0 - a6207cadad0acd1876f436dc6baeddf46c42af06

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
build using command line python 3.7.5

#### How is this defect reproduced ? 
Build the code for any USB device example and flash the bin file to LPC17xx target.

0xc0170, 2020-06-30:
> After investigating, in file targets/TARGET_NXP/USBHAL_LPC17xx.cpp, line 414, ThisThread::sleep_for is causing this issue.
When i replaced that line of code with wait_us it works.

```
    osStatus_t status = osDelay(rel_time.count());
    MBED_ASSERT(status == osOK);
```

Can you check why this assert is failing ? I believe it is because the sleep_for() is executed from ISR, so an error is returned.

 Where is `targets/TARGET_NXP/USBHAL_LPC17xx.cpp` ? I can't find it in Mbed OS.

vvkaggarwal12, 2020-06-30:
> > After investigating, in file targets/TARGET_NXP/USBHAL_LPC17xx.cpp, line 414, ThisThread::sleep_for is causing this issue.
> > When i replaced that line of code with wait_us it works.
> 
> ```
>     osStatus_t status = osDelay(rel_time.count());
>     MBED_ASSERT(status == osOK);
> ```
> 
> Can you check why this assert is failing ? I believe it is because the sleep_for() is executed from ISR, so an error is returned. or IS is masked (most probably). This is the function executing `USBPhyHw::init(USBPhyEvents *events)`. I only see there `NVIC_DisableIRQ(USB_IRQn);`. Please check what is being returned and if ISR are really disabled.

sleep_for() is not being called from ISR. As you said, it is being called from USBPhyHw::init(USBPhyEvents *events). Please check https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_NXP/USBHAL_LPC17.cpp line number 414.

0xc0170, 2020-06-30:
Yes, but dont know the app flow thus would check the callstack. The best would be to step into sleep_for -> osDelay and find out why it returns error - is it because interrupts are masked or any active interrupt ? Run a debug session to find out.

vvkaggarwal12, 2020-06-30:
> Yes, but dont know the app flow thus would check the callstack. The best would be to step into sleep_for -> osDelay and find out why it returns error - is it because interrupts are masked or any active interrupt ? Run a debug session to find out.

It is because the interrupts are masked. Don't have idea from where the interrupts are getting masked during this process.

0xc0170, 2020-06-30:
```
void USBDevice::init()
{
    lock();

    if (!_initialized) {
        this->_phy->init(this);
        _max_packet_size_ep0 = this->_phy->ep0_set_max_packet(MAX_PACKET_SIZE_EP0);
        _initialized = true;
    }

    unlock();
}
```

correct, just found it. `sleep_for()` should not be there as it is not suitable. As this is very short timeout, can you send a pull request using wait_us instead? that fixes it as you stated earlier. Thanks

vvkaggarwal12, 2020-06-30:
> ```
> void USBDevice::init()
> {
>     lock();
>
>     if (!_initialized) {
>         this->_phy->init(this);
>         _max_packet_size_ep0 = this->_phy->ep0_set_max_packet(MAX_PACKET_SIZE_EP0); [...]

Thank you very much for the help.
Yes sure i will give a pull request for this.

PR #13206 (2020-07-01): Usb device init: wait_us fix

### Summary of changes 
"fixes #13203"
In usb device phy initialization, ThisThread::sleep_for changed to wait_us.
As before the usb phy is initialized, all the interrupts are disabled, so sleep_for is not supported.
USB device was not getting enumerated as a result of this delay.
Calling wait_us will resolve the issue.

#### Impact of changes 

#### Migrati
commit 8d1503b2 (2020-07-01): Fix: #13203: USB device: wait_us instead of ThisThread::sleep_for, delay given is 5us
closed 2020-07-01 by 0xc0170