ARMmbed/mbed-os #13208: arm_math.h is still there, but CMSIS-DSP is gone

janjongboom, 2020-06-30:
### Description of defect

CMSIS-DSP has been removed from Mbed OS with the release of Mbed OS 6, but the `arm_math.h` header is still present, leading to an error when linking when using a CMSIS-DSP version. See f.e. https://github.com/arduino/ArduinoCore-mbed/issues/7 of this in action. Should this function not be removed?

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

Mbed OS 6.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

n/a

#### How is this defect reproduced ? 

n/a

0xc0170, 2020-07-01:
From what I recall, the functionality of that header was used as it was part of the cmsis-core not dsp (that was some time ago, it is now part of dsp https://github.com/ARM-software/CMSIS_5/blob/develop/CMSIS/DSP/Include/arm_math.h). The header stayed for backward compatibility. It should have been removed in the latest major version, anyway, can be fixed.

I can try to remote it and see how many things fall apart.

janjongboom, 2020-07-01:
@0xc0170 Yeah, the header is out of sync as well. E.g. `arm_q15_to_float` signature is incorrect with latest CMSIS-DSP.

0xc0170, 2020-07-07:
I searched within Mbed OS, nothing includes the header. However we cant remove it now without introducing a breaking change.

What I would do, as a workaround: define `_ARM_MATH_H` in the global scope. This macro definition would skip the entire file and you can use upstream math repository.