ARMmbed/mbed-os #13211: Serial: error: unknown type name and waiterror: use of undeclared identifier

JohanScheepers, 2020-07-01:
### Description of defect

I am compiling for a mDot board - have a problem with the Serial and wait

Building project mDot_AnalogIn_Example (MTS_MDOT_F411RE, ARMC6)
Scan: mDot_AnalogIn_Example
Compile [  1.7%]: CAN.cpp
Compile [  1.8%]: BusOut.cpp
Compile [  2.0%]: BusInOut.cpp
Compile [  2.1%]: main.cpp
[Warning] MTSLog.h@14,12: returning address of local temporary object [-Wreturn-stack-address]
[Error] mDot.h@1769,30: no member named 'attach' in 'mbed::Callback<void ()>'
[Error] main.cpp@29,1: unknown type name 'Serial'
[Error] main.cpp@48,17: use of undeclared identifier 'wait'
[Error] main.cpp@62,13: use of undeclared identifier 'wait_ms'
[Error] main.cpp@64,13: use of undeclared identifier 'wait_ms'
[Error] main.cpp@199,13: use of undeclared identifier 'wait_ms'
[ERROR] In file included from .\main.cpp:2:
In file included from ./libmDot-mbed5\mDot.h:34:
In file included from ./libmDot-mbed5/Mote.h:22:
In file included from ./libmDot-mbed5/MacEvents.h:19:
In file included from ./libmDot-mbed5/Lora.h:23:
./libmDot-mbed5/MTS-Lora/vendor/multitech/MTS-Utils\MTSLog.h:14:12: warning: returning address of local temporary object [-Wreturn-stack-address]
return prettyFunction.substr(begin,end).c_str();
^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
In file included from .\main.cpp:2:
./libmDot-mbed5\mDot.h:1769:30: error: no member named 'attach' in 'mbed::Callback<void ()>'
_wakeup_callback.attach(object, member);
~~~~~~~~~~~~~~~~ ^
.\main.cpp:29:1: error: unknown type name 'Serial'
Serial pc(USBTX, USBRX);
^
.\main.cpp:48:17: error: use of undeclared identifier 'wait'
wait(delay_s);
^
.\main.cpp:62:13: error: use of undeclared identifier 'wait_ms'
wait_ms(mDot::getInstance()->getNextTxMs());
^
.\main.cpp:64:13: error: use of undeclared identifier 'wait_ms'
wait_ms(1000);
^
.\main.cpp:199:13: error: use of undeclared identifier 'wait_ms'
wait_ms(10);
^
1 warning and 6 errors generated.

-->

#### Target(s) affected by this defect ?
All

#### Toolchain(s) (name and version) displaying this defect ?
All

#### What version of Mbed-os are you using (tag or sha) ?

Mbed Studio: 1.0.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed Studio: 1.0.0

#### How is this defect reproduced ? 

Can not compile

JohanScheepers, 2020-07-01:
Hi,

Sorry new to mbed, where do you finf sha?

0xc0170, 2020-07-01:
What application do you use?  How can we reproduce? There is not enough details provided.

Do you include "mbed.h" ?

0xc0170, 2020-07-01:
> #### What version of Mbed-os are you using (tag or sha) ?

This should be git sha or tag (mbed-os-6.0.0 for instance)

JohanScheepers, 2020-07-01:
#a6207cadad0acd1876f436dc6baeddf46c42af06

#include "mbed.h"
#include "mDot.h"
#include "ChannelPlans.h"
#include "MTSLog.h"
#include "MTSText.h"

Target
Multitech mDot (MTS_MDOT_F411RE)

JohanScheepers, 2020-07-01:
So now I have tried different versions. Same code

Works on https://github.com/ARMmbed/mbed-os.git#64853b354fa188bfe8dbd51e78771213c7ed37f7

Does not work on - https://github.com/ARMmbed/mbed-os.git#a6207cadad0acd1876f436dc6baeddf46c42af06

https://github.com/ARMmbed/mbed-os/#a6207cadad0acd1876f436dc6baeddf46c42af06

JohanScheepers, 2020-07-01:
I have tried the same code on the following

Works on - https://github.com/ARMmbed/mbed-os.git#64853b354fa188bfe8dbd51e78771213c7ed37f7

Does not work on - 
https://github.com/ARMmbed/mbed-os/#a6207cadad0acd1876f436dc6baeddf46c42af06
https://github.com/ARMmbed/mbed-os.git#a6207cadad0acd1876f436dc6baeddf46c42af06

JohanScheepers, 2020-07-01:
You need the the following lib

libmDot-mbed5
mbed-os

And the code below in the main.cpp

#include "mbed.h"
#include "mDot.h"
#include "ChannelPlans.h"
#include "MTSLog.h"
#include "MTSText.h"

// Set GPIO1 for AnalogIn (used to measure analog voltage)
AnalogIn tVin(GPIO1);

 
// Create variables with settings you want.
static std::string network_name = "xxxxxx";
static std::string network_passphrase = "xxxxxx";
static uint8_t frequency_sub_band = 0;
static lora::NetworkType network_type = lora::PUBLIC_LORAWAN;
static uint8_t join_delay = 5;
static uint8_t ack = 0;
static bool adr = true;

// mDot Object
mDot* dot = NULL;
// Channel Plan object
lora::ChannelPlan* plan = NULL;

// Setup the debug port
Serial pc(USBTX, USBRX);

void join_network() {
    int32_t j_attempts = 0;
    int32_t ret = mDot::MDOT_ERROR;
    
    // attempt to join the network
    while (ret != mDot::MDOT_OK) {
        logInfo("attempt %d to join network", ++j_attempts);
        logInfo("do Join attempt");
        ret = dot->joinNetwork();
        logInfo("Join attempt done");
        if (ret != mDot::MDOT_OK) {
            logError("failed to join network %d:%s", ret, mDot::getReturnCodeString(ret).c_str());
            // in some frequency bands we need to wait until another channel is available before transmitting again
            uint32_t delay_s = (dot->getNextTxMs() / 1000) + 1;
            if (delay_s < 5) {
                logInfo("waiting %lu s until next free channel", delay_s);
                wait(delay_s);
            } else {
                logInfo("sleeping %lu s until next free channel", delay_s);
                dot->sleep(delay_s, mDot::RTC_ALARM, false);
            }
        }
    }
}

void wait_to_send() {
    while(!mDot::getInstance()->getIsIdle() || 
          mDot::getInstance()->getNextTxMs() != 0 ) {

        if (mDot::getInstance()->getNextTxMs() > 1000)
            wait_ms(mDot::getInstance()->getNextTxMs());
        else
            wait_ms(1000);
    }
}

[... 112 lines cut ...]
        tx_data.push_back(d & 0xFF);
        logInfo("Sending Float as 4 bytes: [0x%08X]", d);
        send_packet(tx_data);
            
        
        // Send Vin as a readable string
        std::string strVin = std::to_string(Vin_Volt);
        logInfo("Sending Voltage as String: %s", strVin.c_str());
        send_string_packet(strVin);
        
                        
        // sleep
        // Wait for mDot to become idle before sleeping
        while (true) {
            if (dot->getIsIdle()) {
                break;
            }
            wait_ms(10);
        }
        logInfo("Sleeping for %d seconds.", sleep_secs);
        dot->sleep(10, mDot::RTC_ALARM, false); // Deepsleep not currently supported on mDot
    }

    return 0;
}

0xc0170, 2020-07-02:
Thanks for details. Some of the functionality was removed in Mbed OS 6 after deprecation. Please read the release notes for Mbed OS 6 to find out more.

To briefly summarize what I remember, use these instead:

`wait_ms` / `wait` -> `ThisThread::sleep_for`
`Serial` -> `BufferedSerial`

0xc0170, 2020-07-02:
I'll close this as resolved, as the errors are related to the changes in Mbed OS 6. Once you update, you should get 0 errors.

closed 2020-07-02 by 0xc0170