ARMmbed/mbed-os #13214: USBMIDI requires attached callback for reading

tschiemer, 2020-07-01:
Without attaching a MIDIEvent callback the internal `_data_ready` is never updated which is set in `_next_message()`.

The order of the conditions in `_out_callback()` should be changed - otherwise `_next_message()` is not called when no callback is set (which would be a strange requirement?).

https://github.com/ARMmbed/mbed-os/blob/ffeb926a673c979047e98b888b44c62450f2d0f2/drivers/source/usb/USBMIDI.cpp#L330

tschiemer, 2020-07-02:
Hi!

I encountered some further misbehaviour with the USBMIDI driver which I tried to resolve here: https://github.com/tschiemer/mbed-os/tree/usbmidi-writeplus-fixes

The suggested change mentioned above is also included.

The additional changes try to remedy several problems:
- BREAKING CHANGE: Renamed static factory Sysex -> Raw (I'm not sure why everything aside from those default messages is considered as SysEx...)
- BREAKING CHANGE: changed signature of `USBMIDI.read(MIDIMessage * m)` to `read(MIDIMessage &m)` and added `read(uint8_t * &data, uint16_t &length, uint16_t maxlength = MAX_MIDI_MESSAGE_SIZE)` (to prevent NULL pointers by default)
- CHANGE: Removed the cable number as the first byte of MIDIMessage -> new field (working with raw data is more intuitive and doesn't require understanding what that first byte is about)
- FIX: `USBMIDI.write(MIDIMessage m)`: the copy-operator leads to hard faults when called in other forms than write(MIDIMessage::NoteOn(1)) (ex. write(&msg)) -> introduced new signatures:
   - `write(MIDIMessage * m)`
   - `write(uint_t * &data, uint16_t length, uint8_t cable_num = CABLE_NUM)`
- FIX: `USBMIDI.write(..)` did not set the USBMIDI Event Packet header correctly (see [USBMIDI specs](https://usb.org/sites/default/files/midi10.pdf))

I can create a pull request if you want, but don't have so much time for additional testing...

If you want I can also add the method signatures to avoid breaking current systems..?

0xc0170, 2020-07-02:
Thanks for the fixes, we will review. As in the another issue, please keep the issue template in the report.

adbridge, 2020-10-15:
As this issue was not updated as requested with the required information I am going to close it. If it is still an issue please raise another one with the correctly filled in fields in the template.

closed 2020-10-15 by adbridge