ARMmbed/mbed-os #13215: STM32 USB device: endpoint_abort not supported

jeromecoutant, 2020-07-01:
### Description of defect

STM32 USB device implementation doesn't currently support endpoint abort procedure.

https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/USBPhy_STM32.cpp#L539

From test point of view, this makes tests-usb_device-basic failed,
specially 2 test cases:
- "endpoint test halt"
- "endpoint test data toggle reset"

@ARMmbed/team-st-mcd 
@MarceloSalazar 

#### Target(s) affected by this defect ?

STM32

#### Toolchain(s) (name and version) displaying this defect ?

NA

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.1.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

NA

#### How is this defect reproduced ? 

NA

jeromecoutant, 2020-07-01:
ST_INTERNAL_REF 47751

agausmann, 2023-02-04:
The latest versions of F7 and H7 HALs do have a [`HAL_PCD_EP_Abort` function](https://github.com/STMicroelectronics/STM32CubeF7/blob/f8bda023e34ce9935cb4efb9d1c299860137b6f3/Drivers/STM32F7xx_HAL_Driver/Inc/stm32f7xx_hal_pcd.h#L360). I haven't checked any of the others, but I imagine that this will make its way to them, if it hasn't already.

It would be great to finally get this fixed! Unfortunate that the stale bot sweeps up legitimate issues that the community could otherwise pick up.