ARMmbed/mbed-os #13218: CY8CKIT_064B0S2_4343W target name over 20 char limit, not matching mbed-os-tools

maclobdell, 2020-07-01:
### Description of defect

The target name for the PSoC64 production board is `CY8CKIT_064B0S2_4343W` in Mbed OS. However, this target name is over the 20 character limit.

`mbed-os-tools`  uses a different target name `CY8CKIT064B0S2_4343W` which is under the limit. 

The mismatch results in at least two problems:
- greentea test infrastructure failing to detect the right board to test
- online compiler and possibly mbed studio can't compile for the right target

#### Target(s) affected by this defect ?

CY8CKIT_064B0S2_4343W

#### Toolchain(s) (name and version) displaying this defect ?

n/a

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.1.0
    
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli           1.10.1
mbed-greentea      1.7.4
mbed-host-tests    1.5.10
mbed-ls            1.7.10
mbed-os-tools      0.0.14

#### How is this defect reproduced ? 

```
mbed test -m CY8CKIT_064B0S2_4343W -t gcc_arm
```
Result
```
mbedgt: no platform/target matching tests were found!
mbedgt: no matching platforms were found!

```

**Workaround for testing**

Use the mock command to change the target name. this solves problems with testing
It does not solve Mbed Studio detect problem or other potential problems.

```
mbedls -m 1910:CY8CKIT_064B0S2_4343W
```

0xc0170, 2020-07-02:
I hope to get this 20 char limits resolved in tools in the future. We can see that introducing more targets that come from the same family chip get easily over 20 chars.

ifyall, 2020-09-18:
This has been addressed in https://github.com/ARMmbed/mbed-os/pull/13516.

MarceloSalazar, 2020-10-02:
Closing as resolved.

closed 2020-10-02 by MarceloSalazar