ARMmbed/mbed-os #13220: Waiting for rtos::EventFlag in USBMIDI.write(..) blocks on NUCLEO-H743ZI2

tschiemer, 2020-07-02:
Inserting a delay `wait_us(100)` just before remedies the problem but this would indicate an underlying problem ..

https://github.com/ARMmbed/mbed-os/blob/ffeb926a673c979047e98b888b44c62450f2d0f2/drivers/source/usb/USBMIDI.cpp#L129

Also see [USBMIDI example](https://os.mbed.com/docs/mbed-os/v6.0/apis/usbmidi.html) to test.

- macOS 10.15.5
- mbed Studio 1.0.0.6 
- [mbed-os.lib](https://github.com/ARMmbed/mbed-os.git#165be79392ae7b1bee4388d2bc8ed8281978f07c)

0xc0170, 2020-07-02:
@tschiemer Please can you keep the issue template? we have a parser to get the data and check, thanks

Can you also add details why it should not block (if it blocks as stated and should not) ?

tschiemer, 2020-07-04:
>@tschiemer Please can you keep the issue template? we have a parser to get the data and check, thanks
Ah sorry - I created a new issue from the code line and not by pressing *New Issue* so i never saw a issue template..

> Can you also add details why it should not block (if it blocks as stated and should not) ?

Essentially the method `USBMIDI::write(MIDIMessage m)` has the goal of sending a MIDIMessage to the connected host.
On said board (NUCLEO-H743ZI2) it blocks indefinitely and the device becomes unresponsive - apart from never sending anything to the host.
This is why it should not block - other than the way actually intended.

The mentioned delay resolves this issue on said board, but this is not an actual fix of the underlying problem, possibly related to https://github.com/ARMmbed/mbed-os/issues/13215 but I have no idea.

havencking, 2021-06-07:
USBMIDI write also blocks and becomes unresponsive with an LPC1768 with the bare-metal profile and bus-powered if the USB host isn't reading the messages/events transmitted. As soon as the host is set to read (i.e., my device is selected by an application as a USB MIDI input), the LPC1768 become responsive again.

I changed osWaitForever to 1, and that seems to make it work.

 uint32_t flags = _flags.wait_any(FLAG_WRITE_DONE | FLAG_DISCONNECT, 1, false); 

I'm using Mbed Studio 1.4.1 in Debian linux with mbed-os-6.11.0