ARMmbed/mbed-os #13226: Long write BLE GATT data from client getting dropped by the Mbed OS GATT server in 5.13

cy-prasanthkumar, 2020-07-03:
### Description of defect
The mbed-os target device is using 5.13.1 Mbed OS and running a GATT server application with custom characteristic UUID. The client BLE device whose MTU size is limited to 23, tries to send a *Long write* GATT data to the server which is longer than 23 bytes. We observed that the client data received at Mbed OS target device is truncated to 18 bytes always and did not receive the subsequent bytes.

#### Target(s) affected by this defect ?
CY8CPROTO_062_4343W

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM (v7.2.1)

#### What version of Mbed-os are you using (tag or sha) ?
**Mbed OS tag/branch:** mbed-os-5.13.1
**Mbed OS SHA:** 5941d1718339116cd12914238ec331c84da3d08f

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
**mbed-cli version:** 1.10.0

#### How is this defect reproduced ? 
Run a BLE GATT server application with custom UUID characteristics registered which can accept string of 33 bytes or more and start advertising. The BLE GATT client scans and connects to the GATT server. Send a "Long write" GATT data from the client longer than 23 bytes. Observe that the client data has truncated and did not receive the full bytes sent by the client on the target device running the GATT server application using Mbed OS.

cy-prasanthkumar, 2020-07-03:
Adding @loverdeg.

0xc0170, 2020-07-07:
cc @ARMmbed/mbed-os-pan

pan-, 2020-07-15:
@paul-szczepanek-arm Can you investigate this ?

paul-szczepanek-arm, 2020-07-17:
This is still present, tracked on https://jira.arm.com/browse/IOTPAN-640 and when fixed on master I will also make a patch for the old version.

paul-szczepanek-arm, 2020-08-19:
PR up: https://github.com/ARMmbed/mbed-os/pull/13449

loverdeg, 2020-11-22:
Did !13449 fix this?

pan-, 2020-11-23:
@loverdeg It should have fixed it.

closed 2020-11-23 by 0xc0170