ARMmbed/mbed-os #13259: Power consuption of NRF52 is to high

mkarn3yru96, 2020-07-08:
### Description of defect
I am getting a current consumption of 7.7mA (average) - and always above 7mA when running code compiled by the mbed online OS in a NRF52832 based custom board, as well as in the NRF52-DK.
I am measuring the current with a 10 Ohm resistor in series with the power supply and an oscilloscope.

It looks like the nrf52 never enters the low power (sleep) mode...

#### Target(s) affected by this defect ?
NRF52-DK, NRF52832, BMD-350

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

https://os.mbed.com/users/mreiscarneiro/code/All_gaps_connectionparams_Low_energy_slo/

[image]

mkarn3yru96, 2020-07-09:
I testes the updated heartrate example for mbed os5 (https://os.mbed.com/docs/mbed-os/v5.15/apis/heartrateservice.html#heartrateservice-example) , in embed studio, and the problem persists both in the nrf52-DK and my custom nrf52832 custom board

0xc0170, 2020-07-10:
cc @ARMmbed/mbed-os-pan

pan-, 2020-07-10:
@mkarn3yru96 Thanks for sharing the code source, it seems like an old version of mbed is being used and event processing is made with the function `waitForEvent`. That is not the optimal way of doing it. I'd suggest the use of an event queue and posting an event into it when the ble stack needs processing as it is shown in our examples: https://github.com/ARMmbed/mbed-os-example-ble/blob/1cadccbeac094824d0b656b66a023e7d73c5e0e4/BLE_GAP/source/main.cpp#L757-L768

mkarn3yru96, 2020-07-14:
@ciarmcom it is edited

jrobeson, 2020-08-08:
maybe this should be folded into https://github.com/ARMmbed/mbed-os/issues/10669

0xc0170, 2020-08-31:
The reminder is correct, mbed 2 is not proper sha or tag as requested in the version being reported. 

> maybe this should be folded into #10669. unless mbed 2 isn't using cordio? In any case, isn't mbed os 2 deprecated in favor of mbed os 6 with the bare metal profile?

The code app is using older nrf libs (their stack). Rather to ask this on the forum, there might be someone who already fixed it in their app. This is using deprecated version of Mbed 2 and also other libraries (nrf lib for instance).

Please review @pan-'s suggestion and try it.

I'll close this as won't fix, we are not able to help but community could.

closed 2020-08-31 by 0xc0170