ARMmbed/mbed-os #13265: RTOS idle hooks overridden by some targets

LDong-Arm, 2020-07-10:
### Description of defect

The Mbed OS default RTOS idle hook supports for Tickless, and targets should not attempt to override that. But a few target does that, and `rtos_attach_idle_hook()` they use is a private function of the RTOS library which shouldn't be called from the outside. 

_Private headers will not be accessible from the outside_ in the new build tool we are working on.

https://github.com/ARMmbed/mbed-os/blob/efe46b5aa17172374c0b40798a025ba6d9101274/targets/TARGET_Cypress/TARGET_PSOC6/mbed_overrides.c#L111-L115
where `active_idle_hook` is an _empty function_.

And

https://github.com/ARMmbed/mbed-os/blob/efe46b5aa17172374c0b40798a025ba6d9101274/targets/TARGET_Samsung/TARGET_SIDK_S5JS100/sleep.c#L71-L82

Both "customised hooks" don't do much - they can probably be dropped.

#### Target(s) affected by this defect ?
* Cypress MCU_PSOC6
* Samsung S5JS100 

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.1.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
N/A

0xc0170, 2020-07-16:
Please review, this should be fixed

@ARMmbed/team-samsung 
@ARMmbed/team-cypress

zhizhezhu, 2020-07-21:
hi
`s5js100_idle_hook()` can dropped, and can use `default_idle_hook()`. Please delete the `if (initialize_policy == 0) {
        set_sleep_policy();
    }` code.

And please explain what the `sleep_manager_lock_deep_sleep()` and `sleep_manager_unlock_deep_sleep()` function？

LDong-Arm, 2020-07-21:
> hi
> `s5js100_idle_hook()` can dropped, and can use `default_idle_hook()`. Please delete the `if (initialize_policy == 0) { set_sleep_policy(); }` code.

Thanks for confirming this.

> 
> And please explain what the `sleep_manager_lock_deep_sleep()` and `sleep_manager_unlock_deep_sleep()` function？

Deep sleep is similar to sleep but saves more power, has a longer wake up time and disables high frequency clocks. Some drivers rely on high frequency clocks (for example), so they can lock deep sleep to prevent it from entering as needed.

For more details see https://os.mbed.com/docs/mbed-os/v6.2/apis/power-management-sleep.html

ifyall, 2020-07-21:
@kyle-cypress, please look at this and comment. I know that we made these changes in order for us to support deep sleep. If we can't use this method, what can we do?

Ian

zhizhezhu, 2020-07-22:
Yes @ifyall , if we use `default_idle_hook()`, the `sleep_manager_lock_deep_sleep()` will lock the sleep mode. The  Deep sleep will not be allowed in the `sleep_manager_sleep_auto()`.

LDong-Arm, 2020-07-22:
> Yes @ifyall , if we use `default_idle_hook()`, the `sleep_manager_lock_deep_sleep()` will lock the sleep mode.

Well, this is not the case for targets that support Tickless, [including all PSOC6 targets](https://github.com/ARMmbed/mbed-os/blob/7592d702b294a3c23cded62c2a14fdd9edba4fde/targets/targets.json#L5865) in Mbed OS.

There are two versions of `default_idle_hook`, and the Tickless version _doesn't_ lock deep sleep:
https://github.com/ARMmbed/mbed-os/blob/7592d702b294a3c23cded62c2a14fdd9edba4fde/rtos/source/TARGET_CORTEX/mbed_rtx_idle.cpp#L133-L154

(Note: the version _before_ `#else  // MBED_TICKLESS` is Tickless - there's a `#ifdef MBED_TICKLESS` at the beginning of the file.)

So, by using the default idle hook on PSOC6 targets, we can take advantage of Tickless and deep sleep (if available), both of which save power.

kyle-cypress, 2020-07-23:
The `CY_CFG_PWR_SYS_IDLE_MODE` is a macro generated by the Cypress graphical configuration tools. In all of the targets that we ship with Mbed, it default to deepsleep, but it could be changed to "Sleep" or "Active" in a custom target. The code cited above is intended to enforce the user's selection by keeping the CPU from entering DeepSleep (if "Sleep" idle mode was selected) or Sleep (if "Active" idle mode was selected).

If idle mode is "Sleep" (never enter DeepSleep), we can prevent going to DeepSleep in a straightforward manner by calling into the sleep manager to lock deepsleep.

However, if the idle mode is "Active" (never enter Sleep or DeepSleep), there is no equivalent function to lock sleep entry. So this is achieved by attaching a custom idle hook that does nothing, so that the CPU spins while idle instead of going to sleep. I would expect that this mode is mostly used for debugging.

If the ability to attach a custom idle hook is going away, is there another officially supported mechanism of keeping the RTOS from trying to go to sleep in idle?

LDong-Arm, 2020-07-23:
> However, if the idle mode is "Active" (never enter Sleep or DeepSleep), there is no equivalent function to lock sleep entry. So this is achieved by attaching a custom idle hook that does nothing, so [...]

To force the CPU active, what about setting the sleep HAL to empty according to that debug macro instead? https://github.com/ARMmbed/mbed-os/blob/1bd138b6fc02f90949a7d1aaccbf6176dfa2c933/targets/TARGET_Cypress/TARGET_PSOC6/cy_sleep_api.c#L27-L30

> If the ability to attach a custom idle hook is going away, is there another officially supported mechanism of keeping the RTOS from trying to go to sleep in idle?

If overriding the sleep HAL implementation isn't sufficient, then we might consider this. Any ideas? @evedon @kjbracey-arm ?

kyle-cypress, 2020-07-24:
I believe that setting the sleep HAL implementation is feasible. It would have slightly different semantics than what we have today in that it would block all sleep attempts rather than just those resulting from the RTOS going idle (although that would be more consistent with what the deepsleep lock does). 
I'm checking with the group that defined CY_CFG_PWR_SYS_IDLE_MODE to confirm this would be consistent with their intent, and will update when I hear back.

LDong-Arm, 2020-07-27:
@kyle-cypress Alternatively, we might be able to disable ("undeclare") sleep in the configuration.
In the `mbed_app.json` of the app we want to build, do
```json
    "target_overrides": {
        "MCU_PSOC6": {
            "target.device_has_remove": ["SLEEP"]
        }
    }
```
this is likely the most straightforward, but I haven't tested it personally.

LDong-Arm, 2020-07-28:
@evedon @MarceloSalazar From the discussion above, it's been confirmed from Samsung side we can remove the idle hook from their targets.

As for Cypress targets, what they want to achieve is bypass sleep with an active (non-sleep) debug macro defined. @kyle-cypress Could you please let us know once the alternative approach has been confirmed (or not), thanks.

Please note the header containing `rtos_attach_idle_hook()` will likely _not_ be available to drivers anymore once we implement stricter paths rules for headers.

0xc0170, 2020-10-22:
@LDong-Arm What is the outcome of this issue? I do not see referenced any PR, but there were proposals above to fix targets.

@kyle-cypress @zhizhezhu Has this been addressed or can you send fixes please?

closed 2020-12-17 by ciarmcom