ARMmbed/mbed-os #13271: LICENSE.md is missing FATFS / ChaN

JanneKiiskila, 2020-07-10:
### Description of defect

FATFS credits / license is missing from https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.15/LICENSE.md.

https://github.com/ARMmbed/mbed-os/tree/mbed-os-5.15/features/storage/filesystem/fat/ChaN

#### Target(s) affected by this defect ?

All using FAT FS.

#### Toolchain(s) (name and version) displaying this defect ?

Irrelevant.

#### What version of Mbed-os are you using (tag or sha) ?

https://github.com/ARMmbed/mbed-os/blob/mbed-os-5.15/LICENSE.md
likely in quite a few versions, though.

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

N/A

#### How is this defect reproduced ? 

Check the LICENSE.md

0xc0170, 2020-07-10:
Thanks for reporting, we shall fix this!

JanneKiiskila, 2020-07-10:
Improvement idea - make a tool, which generates the complete 3rd party IP/License information based on the configuration of the build. Right now it's a lot of manual labour.

JanneKiiskila, 2020-07-10:
https://github.com/ARMmbed/mbed-os/pull/13275 for 5.15.

JanneKiiskila, 2020-07-16:
Holiday seasons on-going...

0xc0170, 2020-07-16:
Thanks @JanneKiiskila 

> Improvement idea - make a tool, which generates the complete 3rd party IP/License information based on the configuration of the build. Right now it's a lot of manual labor.

scandcode should be able to get this... Once we have components in (restructure), we shall look at this and run scancode per component to get a view on licenses.

closed 2020-07-21 by JanneKiiskila