ARMmbed/mbed-os #13292: target linting tool is outdated

JojoS62, 2020-07-14:
### Description of defect

1. The documentation points to a lint.py for checking valid target definitions, but the output does not look reasonable or outdated.
2. The script is also helpful for checking custom targets, but cannot applied to a custom_targets.json

A workaround for the 2nd issue is to create a test brand and add the testing target to targets.json.

E.g. USTICKER, STDIO_MESSAGES and others are allowed but not in the whitelist.
c_lib, public, device_has are inherited but marked as missing.

example output for a custom target base on MCU_STM32:

```
D:\Projects\Sn\LocalGit\Mbed5-Projects\mbed-os>python tools/targets/lint.py targets STM32F407VE_BLACK 
hierarchy: Family (MCU_STM32) -> MCU (STM32F407VE_BLACK) -> ???
hierarchy errors:
- no boards found in hierarchy
target errors:
  MCU_STM32:
  - supported_c_libs found, and is not allowed
  - supported_application_profiles found, and is not allowed
  - macros found, and is not allowed
  - overrides found, and is not allowed
  - _from_file found, and is not allowed
  - USTICKER is not allowed in device_has
  - STDIO_MESSAGES is not allowed in device_has
  - WATCHDOG is not allowed in device_has
  - RESET_REASON is not allowed in device_has
  STM32F407VE_BLACK:
  - macros_add found, and is not allowed
  - overrides found, and is not allowed
  - components_add found, and is not allowed
  - _from_file found, and is not allowed
  - c_lib not found, and is required
  - public not found, and is required
  - device_has not found, and is required
```

the target definition looks like this:

```
    "STM32F407VE_BLACK": {
        "inherits": ["MCU_STM32"],
        "core": "Cortex-M4F",
        "extra_labels_add": [
            "STM32F4",
            "STM32F407xE"
        ],
        "macros_add": [
            "STM32F407xx"
        ],
        "supported_toolchains": ["ARM", "GCC_ARM", "IAR"],
        "config": {
            "clock_source": {
                "help": "Mask value : USE_PLL_HSE_EXTC (need HW patch) | USE_PLL_HSE_XTAL | USE_PLL_HSI",
                "value": "USE_PLL_HSE_XTAL|USE_PLL_HSI",
                "macro_name": "CLOCK_SOURCE"
            }
        },
        "device_has_remove": [
            "SERIAL_FC"
        ],
        "overrides": {
            "lse_available": 0,
            "network-default-interface-type": "ETHERNET"
        },
        "device_has_add": [
            "ANALOGOUT", 
            "MPU", 
            "TRNG", 
            "FLASH",
            "EMAC",
            "SDIO",
            "USBDEVICE"
        ],
        "components_add": ["SDIO", "SPIF"],
        "release_versions": ["2", "5"],
        "device_name": "STM32F407VETx"
    }
```

#### Target(s) affected by this defect ?

all targets in targets.json

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.1.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.2

#### How is this defect reproduced ? 

run 'python tools/targets/lint.py targets NUCLEO_F103RB' or some other target

0xc0170, 2020-07-16:
@JojoS62 Thanks for reporting. The target.json has changed slightly for Mbed 6, but tools like lint there has not been. This is mainly due to our work on updating tools. We shall consider adding something similar to lint.py script.

If this is a simple fix, please send a new pull request.

@ARMmbed/mbed-os-tools Please review

JojoS62, 2020-07-16:
I think also this tool is usefull, especially for testing custom targets. Some settings are easy to add to the whitelist, but for some others its hard to figure out what is valid or not.
For the things that I find I can send a PR.

talorion, 2020-08-18:
I was having similar issues so I started working on the problems.
first I fixed the problem with the custom_targets_json...
I created a PR but it failed in the CI: https://github.com/ARMmbed/mbed-os/pull/13448
it seems that the tools can not be maintained anymore since they have been rewritten, to bad that I did not look at the blog sooner :-)
https://os.mbed.com/blog/entry/Introducing-the-new-Mbed-Tools/
I have to take a look but hope that the problems are solved

JojoS62, 2020-08-18:
yes, I read that announcement also and missed to close this issue. I think also it is neccessary to check when the new tools are released.
There should be more focus on custom targets because you cannot use the eval boards in commercial products. And maintaining custom targets is annoying because of frequent changes in the target.json.

closed 2020-08-18 by JojoS62