ARMmbed/mbed-os #13295: CY8CPROTO_062_4343W failed at cordio-tests-cordio_hci-driver Greentea Test

jamesbeyond, 2020-07-15:
### Description of defect
CY8CPROTO_062_4343W failed at cordio-tests-cordio_hci-driver Greentea Test

The bug was introduced around 11 May - 12 May
because 
af358d150a07d36629581b34f106f1c7a55d1d7b -> PASSED May - 8

296961de027b4dd1cd86fe87e3f3e4bdc9072f68 -> FAILED May -12

#### Target(s) affected by this defect ?
CY8CPROTO_062_4343W

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM and ARMC6

#### What version of Mbed-os are you using (tag or sha) ?
Latest master or 
296961de027b4dd1cd86fe87e3f3e4bdc9072f68 

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
latest mbed-cli/greeentea

#### How is this defect reproduced ? 

mbed test -t GCC_ARM -m CY8CPROTO_062_4343W -n *cordio_hci-driver

jamesbeyond, 2020-07-15:
Hi @pan- Do you have any idea about this ?

0xc0170, 2020-07-16:
@ARMmbed/team-cypress Please review

kyle-cypress, 2020-07-16:
This is likely the same issue as was raised in #13076 and #13075 . We are investigating the failure.