ARMmbed/mbed-os #13297: AC6 LTO Symbol $Sub$$ *all the printf* was not preserved by the LTO

sparker-arm, 2020-07-15:
Hi,

I'm trying to build a selection of examples, including blinky, for benchmarking AC6 with LTO and all the examples fail with symbol errors. I'm using mbed-os@6.0.0 and mbed-cli@1.10.4 and, admittedly, development versions of armclang and armlink... I've seen various patches and fixes around LTO (many thanks!) but I'm wondering whether this is related to https://github.com/ARMmbed/mbed-os/pull/12712? If I understand correctly, minimal-printf is enabled by default now? Is armlink + LTO + minimal printf tested or am I just likely missing some config?

List of linker errors:
Error: L6137E: Symbol $Sub$$_printf_f was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_longlong_oct was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_truncate_signed was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_longlong_hex was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_i was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$vsprintf was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$vsnprintf was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_int_dec was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_x was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_fp_dec was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_wctomb was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_s was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_str was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_charcount was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_o was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_p was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$__2snprintf was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_u was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_truncate_unsigned was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$vprintf was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_c was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_d was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_lc was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$_printf_n was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$__2printf was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$__2sprintf was not preserved by the LTO codegen but is needed by the image.
Error: L6137E: Symbol $Sub$$vfprintf was not preserved by the LTO codegen but is needed by the image.

Thanks,
Sam

### Description of defect

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2020-07-16:
@ARMmbed/mbed-os-core please review

evedon, 2020-07-21:
You are correct: minimal-printf is enabled by default in Mbed OS 6 and this is causing issue with the ARMC6 LTO option because the printf function wrappers are removed by the compiler. A ticket was raised with Arm Compiler team, support case created: CAS-143565-M5Q0F6

sparker-arm, 2020-08-03:
Okay, thanks. Closing as this should be fixed within the linker.

closed 2020-08-03 by sparker-arm