ARMmbed/mbed-os #13299: Target UNO_91H does not compile on mbed-os-5.15.4

observator-x, 2020-07-15:
[Error] rfbits.h@53,0:  #20: identifier "TestPins" is undefined
[Error] rfbits.h@54,0:  #20: identifier "Se2435Pins" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@276,0:  #20: identifier "TEST_CSMA_DONE" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@277,0:  #65: expected a ";"
[Error] NanostackRfPhyAT86RF215.cpp@500,0:  #20: identifier "TEST_CSMA_DONE" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@501,0:  #65: expected a ";"
[Error] NanostackRfPhyAT86RF215.cpp@543,0:  #20: identifier "TEST_CSMA_STARTED" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@544,0:  #159: declaration is incompatible with previous "rf_unlock" (declared at line 203)
[Error] NanostackRfPhyAT86RF215.cpp@550,0:  #20: identifier "TEST_CSMA_STARTED" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@551,0:  #159: declaration is incompatible with previous "rf_unlock" (declared at line 203)
[Error] NanostackRfPhyAT86RF215.cpp@557,0:  #20: identifier "TEST_CSMA_DONE" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@558,0:  #159: declaration is incompatible with previous "rf_backup_timer_stop" (declared at line 97)
[Error] NanostackRfPhyAT86RF215.cpp@588,0:  #20: identifier "TEST_CSMA_STARTED" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@589,0:  #65: expected a ";"
[Warning] NanostackRfPhyAT86RF215.cpp@558,0:  #177-D: function "rf_backup_timer_stop" was declared but never referenced
[Error] NanostackRfPhyAT86RF215.cpp@599,0:  #20: identifier "TEST_TX_DONE" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@600,0:  #18: expected a ")"
[Error] NanostackRfPhyAT86RF215.cpp@613,0:  #20: identifier "TEST_TX_STARTED" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@614,0:  #18: expected a ")"
[Error] NanostackRfPhyAT86RF215.cpp@636,0:  #20: identifier "TEST_RX_DONE" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@637,0:  #159: declaration is incompatible with previous "rf_backup_timer_stop" (declared at line 97)
[Error] NanostackRfPhyAT86RF215.cpp@692,0:  #20: identifier "TEST_RX_STARTED" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@701,0:  #20: identifier "TEST_RX_DONE" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@702,0:  #159: declaration is incompatible with previous "rf_lock" (declared at line 198)
[Error] NanostackRfPhyAT86RF215.cpp@938,0:  #20: identifier "TEST_CSMA_DONE" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@939,0:  #65: expected a ";"
[Error] NanostackRfPhyAT86RF215.cpp@941,0:  #20: identifier "TEST_TX_DONE" is undefined
[Error] NanostackRfPhyAT86RF215.cpp@943,0:  #65: expected a ";"
[Error] NanostackRfPhyAT86RF215.cpp@1200,0:  #147: declaration is incompatible with "int RFBits::init_215_driver(RFBits *, <error-type> *, const std::uint8_t *, std::uint8_t *)" (declared at line 53 of "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\rfbits.h")
[Error] NanostackRfPhyAT86RF215.cpp@1224,0:  #147: declaration is incompatible with "int RFBits::init_se2435_pa(<error-type> *)" (declared at line 54 of "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\rfbits.h")
[Warning] NanostackRfPhyAT86RF215.cpp@168,0:  #550-D: variable "test_pins" was set but never used
[DEBUG] Return: 1
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\rfbits.h", line 53: Error:  #20: identifier "TestPins" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\rfbits.h", line 54: Error:  #20: identifier "Se2435Pins" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 276: Error:  #20: identifier "TEST_CSMA_DONE" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 277: Error:  #65: expected a ";"
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 500: Error:  #20: identifier "TEST_CSMA_DONE" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 501: Error:  #65: expected a ";"
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 543: Error:  #20: identifier "TEST_CSMA_STARTED" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 544: Error:  #159: declaration is incompatible with previous "rf_unlock" (declared at line 203)
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 550: Error:  #20: identifier "TEST_CSMA_STARTED" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 551: Error:  #159: declaration is incompatible with previous "rf_unlock" (declared at line 203)
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 557: Error:  #20: identifier "TEST_CSMA_DONE" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 558: Error:  #159: declaration is incompatible with previous "rf_backup_timer_stop" (declared at line 97)
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 588: Error:  #20: identifier "TEST_CSMA_STARTED" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 589: Error:  #65: expected a ";"
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 558: Warning:  #177-D: function "rf_backup_timer_stop" was declared but never referenced
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 599: Error:  #20: identifier "TEST_TX_DONE" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 600: Error:  #18: expected a ")"
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 613: Error:  #20: identifier "TEST_TX_STARTED" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 614: Error:  #18: expected a ")"
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 636: Error:  #20: identifier "TEST_RX_DONE" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 637: Error:  #159: declaration is incompatible with previous "rf_backup_timer_stop" (declared at line 97)
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 692: Error:  #20: identifier "TEST_RX_STARTED" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 701: Error:  #20: identifier "TEST_RX_DONE" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 702: Error:  #159: declaration is incompatible with previous "rf_lock" (declared at line 198)
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 938: Error:  #20: identifier "TEST_CSMA_DONE" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 939: Error:  #65: expected a ";"
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 941: Error:  #20: identifier "TEST_TX_DONE" is undefined
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 943: Error:  #65: expected a ";"
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 1200: Error:  #147: declaration is incompatible with "int RFBits::init_215_driver(RFBits *, <error-type> *, const std::uint8_t *, std::uint8_t *)" (declared at line 53 of "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\rfbits.h")
[DEBUG] Output: "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 1224: Error:  #147: declaration is incompatible with "int RFBits::init_se2435_pa(<error-type> *)" (declared at line 54 of "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\rfbits.h")
[... cut ...]
"mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 1224: Error:  #147: declaration is incompatible with "int RFBits::init_se2435_pa(<error-type> *)" (declared at line 54 of "mbed-os\components\802.15.4_RF\atmel-rf-driver\source\rfbits.h")
"mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp", line 168: Warning:  #550-D: variable "test_pins" was set but never used
mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp: 2 warnings, 29 errors

observator-x, 2020-07-15:
**Target(s) affected by this defect ?**
?

**Toolchain(s) (name and version) displaying this defect ?**
GCC_ARM

**What version of Mbed-os are you using (tag or sha) ?**
mbed-os-5.15.4

**What version(s) of tools are you using. List all that apply (E.g. mbed-cli)**
mbed-cli 1.10.4

**How is this defect reproduced ?**
targets UNO_91H

0xc0170, 2020-07-16:
@observator-x Is this building blinky?

observator-x, 2020-07-16:
No, it wont compile blinky or even an empty project, on 5.12.3 it compiles without errors and works just normal.

observator-x, 2020-07-16:
@0xc0170 No, it wont build blinky or even an empty project, on 5.12.3 it compiles without errors and works just normal.

oclyke, 2020-07-24:
Ran into a similar issue while trying to add SPI support for my target.

```
Compile [  0.6%]: AnalogIn.cpp
Compile [  0.7%]: AnalogOut.cpp
Compile [  0.8%]: at24mac_s2lp.cpp
Compile [  0.9%]: NanostackRfPhyAT86RF215.cpp
[Error] rfbits.h@53,38: 'TestPins' has not been declared
[Error] rfbits.h@54,24: 'Se2435Pins' has not been declared
[Warning] NanostackRfPhyAT86RF215.cpp@195,43: 'int mbed::TimerBase::read_us() const' is deprecated: Use the Chrono-based elapsed_time method.  If integer microseconds are needed, you can use `elapsed_time().count()` [since mbed-os-6.0.0] [-Wdeprecated-declarations]
[Error] NanostackRfPhyAT86RF215.cpp@276,17: 'TEST_CSMA_DONE' was not declared in this scope
[Error] NanostackRfPhyAT86RF215.cpp@500,9: 'TEST_CSMA_DONE' was not declared in this scope
[Warning] NanostackRfPhyAT86RF215.cpp@542,74: 'void mbed::TickerBase::attach_us(mbed::Callback<void()>, us_timestamp_t)' is deprecated: Pass a chrono duration, not an integer microsecond count. For example use `10ms` rather than `10000`. [since mbed-os-6.0.0] [-Wdeprecated-declarations]
[Error] NanostackRfPhyAT86RF215.cpp@543,13: 'TEST_CSMA_STARTED' was not declared in this scope
[Warning] NanostackRfPhyAT86RF215.cpp@549,55: 'void mbed::TickerBase::attach_us(mbed::Callback<void()>, us_timestamp_t)' is deprecated: Pass a chrono duration, not an integer microsecond count. For example use `10ms` rather than `10000`. [since mbed-os-6.0.0] [-Wdeprecated-declarations]
[Error] NanostackRfPhyAT86RF215.cpp@550,5: 'TEST_CSMA_STARTED' was not declared in this scope
[Error] NanostackRfPhyAT86RF215.cpp@557,5: 'TEST_CSMA_DONE' was not declared in this scope
[Warning] NanostackRfPhyAT86RF215.cpp@587,74: 'void mbed::TickerBase::attach_us(mbed::Callback<void()>, us_timestamp_t)' is deprecated: Pass a chrono duration, not an integer microsecond count. For example use `10ms` rather than `10000`. [since mbed-os-6.0.0] [-Wdeprecated-declarations]
[Error] NanostackRfPhyAT86RF215.cpp@588,13: 'TEST_CSMA_STARTED' was not declared in this scope
[Error] NanostackRfPhyAT86RF215.cpp@599,5: 'TEST_TX_DONE' was not declared in this scope
[Error] NanostackRfPhyAT86RF215.cpp@613,5: 'TEST_TX_STARTED' was not declared in this scope
[Error] NanostackRfPhyAT86RF215.cpp@636,5: 'TEST_RX_DONE' was not declared in this scope
[Error] NanostackRfPhyAT86RF215.cpp@692,5: 'TEST_RX_STARTED' was not declared in this scope
[Error] NanostackRfPhyAT86RF215.cpp@701,5: 'TEST_RX_DONE' was not declared in this scope
[Error] NanostackRfPhyAT86RF215.cpp@938,9: 'TEST_CSMA_DONE' was not declared in this scope
[Error] NanostackRfPhyAT86RF215.cpp@941,5: 'TEST_TX_DONE' was not declared in this scope
[Warning] NanostackRfPhyAT86RF215.cpp@963,60: 'void mbed::TickerBase::attach_us(mbed::Callback<void()>, us_timestamp_t)' is deprecated: Pass a chrono duration, not an integer microsecond count. For example use `10ms` rather than `10000`. [since mbed-os-6.0.0] [-Wdeprecated-declarations]
[Error] NanostackRfPhyAT86RF215.cpp@1200,5: no declaration matches 'int RFBits::init_215_driver(RFBits*, TestPins*, const uint8_t*, uint8_t*)'
[Error] NanostackRfPhyAT86RF215.cpp@1224,5: no declaration matches 'int RFBits::init_se2435_pa(Se2435Pins*)'
[Warning] NanostackRfPhyAT86RF215.cpp@208,15: 'int8_t rf_device_register(const uint8_t*)' defined but not used [-Wunused-function]
[Warning] NanostackRfPhyAT86RF215.cpp@168,18: 'test_pins' defined but not used [-Wunused-variable]
[ERROR] In file included from .\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp:166:
.\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\rfbits.h:53:38: error: 'TestPins' has not been declared
     int init_215_driver(RFBits *_rf, TestPins *_test_pins, const uint8_t mac[8], uint8_t *rf_part_num);
                                      ^~~~~~~~
.\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\rfbits.h:54:24: error: 'Se2435Pins' has not been declared
     int init_se2435_pa(Se2435Pins *_se2435_pa_pins);
                        ^~~~~~~~~~
.\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp: In function 'uint32_t rf_get_timestamp()':
.\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp:195:43: warning: 'int mbed::TimerBase::read_us() const' is deprecated: Use the Chrono-based elapsed_time method. 
 If integer microseconds are needed, you can use `elapsed_time().count()` [since mbed-os-6.0.0] [-Wdeprecated-declarations]
     return (uint32_t)rf->tx_timer.read_us();
                                           ^
In file included from .\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp:30:
./mbed-os/drivers/Timer.h:93:9: note: declared here
     int read_us() const;
         ^~~~~~~
.\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp: In function 'int8_t rf_extension(phy_extension_type_e, uint8_t*)':
.\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp:276:17: error: 'TEST_CSMA_DONE' was not declared in this scope
                 TEST_CSMA_DONE
                 ^~~~~~~~~~~~~~
.\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp:276:17: note: suggested alternative: 'CCA_DONE'
                 TEST_CSMA_DONE
                 ^~~~~~~~~~~~~~
                 CCA_DONE
.\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp: In function 'void rf_csma_ca_timer_interrupt()':
[... cut ...]
.\mbed-os\components\802.15.4_RF\atmel-rf-driver\source\NanostackRfPhyAT86RF215.cpp:168:18: warning: 'test_pins' defined but 
not used [-Wunused-variable]
 static TestPins *test_pins;                                                                                                 not used [-Wunused-variable]
                  ^~~~~~~~~

```

My repo is behind by several commits including this one which relocated some RF drivers:
https://github.com/ARMmbed/mbed-os/commit/f4c62f6369fdf6cd287fcd48733e49ecc2027ea3#diff-4725dcf446a342f1473a8228e42dfa48

This could be part of the issue

oclyke, 2020-07-24:
More observations:
* Issue does not manifest when ```NanostackRfPhyAT86RF215``` is excluded by preprocessor (e.g. ```DEVICE_SPI``` is not defined)
* ```class TestPins;``` is forward-declared in ```NanostackRfPhyAtmel.h```
* ```class TestPins{}``` is defined in ```rfbits.h``` *after* being used as a parameter type in the ```RFBits``` class ```init_215_driver``` method.

```rfbits.h``` needs a forward declaration of ```TestPins``` for this to be valid, but none of the included headers seem like likely candidates to include ```NanostackRfPhyAtmel.h```

``` c++
#include "DigitalIn.h"
#include "DigitalOut.h"
#include "InterruptIn.h"
#include "SPI.h"
#include <Timer.h>
#include "Timeout.h"
#include "rtos.h"
```

I am left wondering who is responsible for declaring ```TestPins```. Is this supposed to be handled by a port? That would seem quite odd IMO.

oclyke, 2020-07-24:
Here's what needs to be fixed:

* ```rfbits.h``` needs a forward declaration of ```TestPins``` - perhaps it could simply include ```NanostackRfPhyAtmel.h```
* ```rfbits.h``` needs a forward declaration of ```Se2435Pins``` - perhaps it could simply include ```NanostackRfPhyAtmel.h```
* the preprocessor guards in ```NanostackRfPhyAtmel.h``` and ```NanostackRfPhyAT86RF215.cpp``` should match
  * ```NanostackRfPhyAtmel.h``` checks:
    * ```#if defined(MBED_CONF_NANOSTACK_CONFIGURATION) && DEVICE_SPI && DEVICE_I2C && defined(MBED_CONF_RTOS_PRESENT)```
  * ```NanostackRfPhyAT86RF215.cpp``` checks 
    * ```#if defined(MBED_CONF_NANOSTACK_CONFIGURATION) && DEVICE_SPI && DEVICE_INTERRUPTIN && defined(MBED_CONF_RTOS_PRESENT)```
  * the differences:
    * source file ensures that device has ```DEVICE_INTERRUPTIN```, header does not
    * header ensure device has ```DEVICE_I2C```, source file does not
  * this means that in some cases the source file can be included without all the necessary declarations
    * in my case ```DEVICE_I2C``` is not defined - causing the header not to be included and these preprocessor statements (among others, most likely) to be missing:
      ``` c++
      #define TEST_TX_STARTED
      #define TEST_TX_DONE
      #define TEST_RX_STARTED
      #define TEST_RX_DONE
      #define TEST_CSMA_STARTED
      #define TEST_CSMA_DONE
      #define TEST_SPARE_1_ON
      #define TEST_SPARE_1_OFF
      #define TEST_SPARE_2_ON
      #define TEST_SPARE_2_OFF
      ```

There may be more items - I am still working toward a successful compilation

talorion, 2020-08-20:
I am having a similar problem compiling tests...
I just created a PR https://github.com/ARMmbed/mbed-os/pull/13471
I think this fixes the problem(at last in my setup it does)

adbridge, 2020-10-15:
@talorion can this now be closed ?

adbridge, 2020-10-27:
As we've had no response I'm going to close this, please open a new defect if the problem persists.

closed 2020-10-27 by adbridge