ARMmbed/mbed-os #13310: IAR export does not set PROJ_DIR option

malavikasajikumar, 2020-07-20:
### Description of defect

When an Mbed project is exported to IAR from the Mbed online compiler (using the right click and Export), looks like the PROJ_DIR option is not set in the resulting IAR project. This is what the C\C++ compiler options looks like for the exported project:

[image]

This lacks the PROJ_DIR option.

In the original Mbed project, if mbed-os is deleted and then manually imported, this problem doesn’t exist. 

What must be causing this difference?

#### Target(s) affected by this defect ?

N/A

#### Toolchain(s) (name and version) displaying this defect ?

IAR

#### What version of Mbed-os are you using (tag or sha) ?

N/A
#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
IAR 8.32.4.

#### How is this defect reproduced ? 

N/A

0xc0170, 2020-07-20:
What error are you getting  ? 
`PROJ_DIR` comes from IAR, the root of the application - my understanding. we use this variable to use relative paths to the project itself.

malavikasajikumar, 2020-07-20:
> What error are you getting ?
> `PROJ_DIR` comes from IAR, the root of the application - my understanding. we use this variable to use relative paths to the project itself.

0xc0170, 2020-07-21:
To compare, how does work with blinky and using `mbed-cli` ? 

I used `mbed export -m K64F -i iar` and I can see in .ewp file the paths set correctly:

```
      <name>mbed-os-example-blinky</name>
      <file> <name>$PROJ_DIR$\mbed_config.h</name> </file>
      <file> <name>$PROJ_DIR$\./main.cpp</name> </file>
      </group>
```

```
                   <name>CCIncludePath2</name>
                    <state>$PROJ_DIR$\.</state>
                    <state>$PROJ_DIR$\./mbed-os</state>
                    <state>$PROJ_DIR$\./mbed-os/cmsis</state>
                    <state>$PROJ_DIR$\./mbed-os/cmsis/TARGET_CORTEX_M</state>
                    <state>$PROJ_DIR$\./mbed-os/connectivity</state>
                    <state>$PROJ_DIR$
```

malavikasajikumar, 2020-07-22:
@0xc0170 That's correct. In Mbed CLI, this is not a problem. Looks like PROJ_DIR option is missing only when exported from the Mbed compiler. Is this something that can be fixed?

0xc0170, 2020-07-27:
> @0xc0170 That's correct. In Mbed CLI, this is not a problem. Looks like PROJ_DIR option is missing only when exported from the Mbed compiler. Is this something that can be fixed?

@ARMmbed/mbed-os-tools Please review

malavikasajikumar, 2020-08-05:
@ARMmbed/mbed-os-tools Hi, could I get an update on this?

0xc0170, 2020-08-13:
https://os.mbed.com/blog/entry/Introducing-the-new-Mbed-Tools/ - as the old tools were frozen, this won't be able to be fixed in the current exporters..

We shall close this as "won't fix" unfortunately.

closed 2020-08-13 by malavikasajikumar