ARMmbed/mbed-os #13319: Cannot run "BLE_PeriodicAdvertising" on STM32WB55RG

eriknayan, 2020-07-20:
### Description of defect

When I try to run this BLE example on a NUCLEO_WB55RG board (which is one of the **supported targets** suggested to be used), I receive this error: `"Periodic advertising not supported, cannot run example.`

It looks like there's no implementation for periodic advertising for this controller, even though it fully supports Bluetooth 5.

#### Target(s) affected by this defect ?

STM32 NUCLEO_WB55RG

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.2.0 -master 3a8989a2479e4f9db227545f29876930a9efe7f7

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.2

#### How is this defect reproduced ? 

Running the `BLE_PeriodicAdvertising` on the target board.

@jeromecoutant will it be implemented soon? Thanks in advance.

jeromecoutant, 2020-08-13:
Hi

Seems that we don't currently support Extended Advertisements, and then Periodic one :-(

Is it mandatory for you ?

Regards,

paul-szczepanek-arm, 2020-08-13:
I think Jerome answered this but just to clarify, periodic advertising is an optional feature of the BLE Spec. So a BLE 5 compliant chip does not guarantee supporting it. All BLE API calls return an error type that includes a value for non implemented features which you need to check.

Ever since 4.2 most features BLE spec adds are optional.

eriknayan, 2020-08-13:
Hi @jeromecoutant  and @paul-szczepanek-arm, thanks for you reply!

Ok, I didn't know that, I thought it was a guaranteed support. Indeed it would be a nice to have feature but we can work around it. 

Thanks again for your help guys.

closed 2020-08-13 by eriknayan