ARMmbed/mbed-os #13324: Bug in memory statistics

lefebvresam, 2020-07-21:
### Description of defect

When using statistics with "MBED_HEAP_STATS_ENABLED=1", the calculation of heap_stats.overhead_size in mbed_alloc_wrappers.cpp is going wrong.

Some tests show that values near the utin32 limit are used in the calculation:
?? + (4294967292 - 24)
and
0 - (4294967292 - 24)

```
void print_heap_statistics() {
    mbed_stats_heap_t heap_stats;
    mbed_stats_heap_get(&heap_stats);
    printf("\nHeap statistics\n");
    printf("---------------\n");
    printf("Bytes allocated currently                      : %lu\n", heap_stats.current_size);
    printf("Max bytes allocated at a given time            : %lu\n", heap_stats.max_size);
    printf("Cumulative sum of bytes ever allocated         : %lu\n", heap_stats.total_size);
    printf("Current number of bytes allocated for the heap : %lu\n", heap_stats.reserved_size);
    printf("Current number of allocations                  : %lu\n", heap_stats.alloc_cnt);
    printf("Number of failed allocations                   : %lu\n", heap_stats.alloc_fail_cnt);
    printf("Overhead added to heap for stats               : %lu\n", heap_stats.overhead_size);
}

Heap statistics
---------------
Bytes allocated currently                      : 10454
Max bytes allocated at a given time            : 10478
Cumulative sum of bytes ever allocated         : 10478
Current number of bytes allocated for the heap : 21264
Current number of allocations                  : 47
Number of failed allocations                   : 0
Overhead added to heap for stats               : 4294966234
```

#### Target(s) affected by this defect ?

NUCLEO_F091RC

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

since mbed 6.2

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

Printing the stats.

lefebvresam, 2020-07-21:
> @lefebvresam thank you for raising this issue.Please take a look at the following comments:
>
> We cannot automatically identify a release based on the version of Mbed OS that you have provided.
> Please [...]

Ticket had been updated.

lefebvresam, 2020-07-21:
Ticket has been updated.

0xc0170, 2020-07-31:
cc @ARMmbed/mbed-os-core please review

PR #13486 (2020-09-03): Fix heap stats overhead_size calculation

### Summary of changes 
`overhead_size` heap stats element was added to maintain real malloc internal overhead size but Mbed OS assumes that `( malloc returned address - 4 )` stores the actual allocated block size( including internal overhead size), but GCC_ARM newlib and newlib nano this location is used for padding when non-alignment memory allocation requested and it stores the negative offset.

So updated Mbed OS allocated block size calculation like when reading from `( malloc returned address - 4 )` is non-negative value consider to be an allocat [... cut]
closed 2020-09-03 by 0xc0170