ARMmbed/mbed-os #13327: External TDB Store Fails to write key on K64F

DMCEngineering, 2020-07-21:
### Description of defect

Configuring the K64F target to use "TDB_EXTERNAL" as its default storage initializes correct but fails when trying to actually write keys. 

#### Target(s) affected by this defect ?
K64F

#### Toolchain(s) (name and version) displaying this defect ?

GCC v9 2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?

mbed version  - #a2ada74770f043aff3e61e29d164a8e78274fcd4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed cli - 1.10.12

#### How is this defect reproduced ? 
app config:
```
{
    "target_overrides": {
        "K64F": {
            "platform.stdio-baud-rate": 115200,
            "storage.storage_type": "TDB_EXTERNAL",
            "storage_tdb_external.external_size": 1048576
        }
    }
}
```
Explicitly limiting the size here of the sd card for speed but using the full size has no impact.

code:
```cpp
int main(void) {

  const char prefix[] = "/kv/";

  int res = kv_reset(prefix);

  if (res) {
    printf("Failed to reset kv: %d", res);
    mbed_die();
  }
  uint32_t write_value = 0;
  uint32_t read_value = 0;
  char key[20] = {0};
  size_t actual_size = 0;

  sprintf(key, "%s%d", prefix, 10);

  int ret = kv_set(key, &write_value, sizeof(write_value), 0);

  if (ret != MBED_SUCCESS) {
    printf("Failed on  error code %d", ret);
    mbed_die();
  }

  ret = kv_get(key, &read_value, sizeof(read_value), &actual_size);

  if (ret != MBED_SUCCESS) {
    printf("Failed read on with error code %d", ret);
    mbed_die();
  }
  MBED_ASSERT(actual_size == sizeof(read_value));
  MBED_ASSERT(read_value == write_value);
}

```

kv_set fails with error code -2130771697

Digging down, the failure seems to be generated from the securestore class: https://github.com/ARMmbed/mbed-os/blob/a2ada74770f043aff3e61e29d164a8e78274fcd4/features/storage/kvstore/securestore/SecureStore.cpp#L258

DMCEngineering, 2020-07-21:
Issue description has been updated.

matthew-macovsky-cypress, 2020-07-29:
I believe that I had the same issue - to resolve it I called ```DeviceKey::get_instance().generate_root_of_trust()``` before calling ```kv_set()```.  See #12921.

DMCEngineering, 2020-07-30:
Confirmed that calling `DeviceKey::get_instance().device_inject_root_of_trust()` does resolve this issue. Might just need a documentation update to clarify the workflow here.

0xc0170, 2020-07-31:
@DMCEngineering as you might know better than me, can you create an issue https://github.com/ARMmbed/mbed-os-5-docs/ with details what was not clear.

DMCEngineering, 2020-08-27:
Closing in favor of documentation update for example here https://github.com/ARMmbed/mbed-os-example-kvstore/issues/52

closed 2020-08-27 by DMCEngineering