ARMmbed/mbed-os #13328: ATCmdParser: can't process uint8_t hex string

pilotak, 2020-07-21:
### Description of defect
When trying to read/parse an AT response (hex string) and assigned to uint8_t array it always fails
TX: `AT$I=11\r\n`
RX: `16141B2F74BA3BB8\r\n`

```cpp
uint8_t test[8];
_parser->send("AT$I=11");

bool ok = _parser->recv("%02hhX%02hhX%02hhX%02hhX%02hhX%02hhX%02hhX%02hhX", &test[0], &test[1], &test[2], &test[3], &test[4], &test[5], &test[6], &test[7]);

if (ok) {
    printf("ok: ");
    for (auto i = 0; i < 8; i++) {
        printf("%02X ", test[i]);
    }

    printf("\n");
} else {
    printf("failed\n");
}
```
Log:
```
AT> AT$I=11
AT? %*02hhX%*02hhX%*02hhX%*02hhX%*02hhX%*02hhX%*02hhX%*02hhX%n
AT< 16141B2F74BA3BB8
AT(Timeout)
```

But if array is set to `unsigned short` it works
```cpp
uint16_t test[8];
_parser->send("AT$I=11");

bool ok = _parser->recv("%02hX%02hX%02hX%02hX%02hX%02hX%02hX%02hX", &test[0], &test[1], &test[2], &test[3], &test[4], &test[5], &test[6], &test[7]);

if (ok) {
    printf("ok: ");
    for (auto i = 0; i < 8; i++) {
        printf("%02X ", test[i]);
    }

    printf("\n");
} else {
    printf("failed\n");
}
```

#### Target(s) affected by this defect ?
all

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

7592d702b294a3c23cded62c2a14fdd9edba4fde

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
 please see above

0xc0170, 2020-07-23:
I am looking at `vrecvscanf` method used internally for `recv()`. It uses `_buffer` that is defined as `char *_buffer` and allocated. Once the receive is done it is stored via `vsscanf` (line 284). 

Have you debugged the `vrecvscanf` if the parsing is correct and later storing it fails or receive already is wrong ?

0xc0170, 2020-07-31:
@pilotak have you found a solution or are able to debug into `vrecvscanf` to find out where the issue is?

pilotak, 2020-07-31:
@0xc0170  sorry, not yet *(because i found a workaround)* i will look at it shortly.

What is my priority is to bring support of NUCLEO_G031K8, could you please look at the PR?