ARMmbed/mbed-os #13329: AnalogIn in Ticker Causes Freezing

MehrdadKhnzd, 2020-07-22:
### Description of defect
When I use AnalogIn in Ticker, my program crashes and it freezes and it starts to blink red led of STM32F3DISCOVERY randomly.

#### Target(s) affected by this defect ?
STM32F3DISCOVERY

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
PlatformIO

#### How is this defect reproduced ? 
```cpp
#include <mbed.h>

Timer timer;

Ticker sampler_ticker;

InterruptIn button(PA_0);

DigitalOut led_blue(PE_8);

AnalogIn temp(ADC_TEMP);

int count = 0;
float data[1000];
bool reading = false;

void sampler() {
  data[(count++ % 1000)] = temp.read();
}

void button_pressed() {
  if (!reading) {
    sampler_ticker.attach_us(&sampler, 400);
    timer.start();
    led_blue = 1;
  } else {
    sampler_ticker.detach();
    timer.stop();
    led_blue = 0;
    count = 0;
  }

  reading = !reading;
}

int main() {
  button.rise(&button_pressed);

  while (1);
}
```

0xc0170, 2020-07-23:
`AnalogIn` class is thread safe , using mutex. Therefore read() in ISR handler is not allowed. 

We wrote older blog post explaining various approaches to the problem (one of them is using events): https://os.mbed.com/blog/entry/Simplify-your-code-with-mbed-events/

Please read and try which one suits your needs.

I'll close this as resolved.

closed 2020-07-23 by 0xc0170

MehrdadKhnzd, 2020-07-23:
Thanks for your response. Some hours ago I found out the solution in another issue and the link was so helpful. Thanks again!