ARMmbed/mbed-os #13339: skipping IRQ handler callbackSerial from UnbufferedSerial

lefebvresam, 2020-07-22:
### Description of defect

The execution of an ISR callback for the UnbufferedSerial that stores each incoming byte in a circular buffer will be skipped after a while. It results in characters hanging in the hardware buffer which will be processed once a new character is received. Experimenting with different approaches did not led to a solution.

I tried:  
* Using priorities like NVIC_SetPriority(USART3_8_IRQn, 255);
* Avoiding find method from std::string: size_t p = command.find("\xFF\xFF\xFF"); and replacing it with a string compare.
* In ISR check for more characters:
```
while(_unbufferedserial.readable()) {... <fill circular buffer> ...}
```

The parts of the code that invoke this behavior:

The constructor implements a thread and an UnbufferedSerial object:
```
CommandInterface::CommandInterface(EventQueue* queue, PinName tx, PinName rx, int baud) 
    : _queue(queue), _unbufferedserial(tx, rx, baud), _rindex(0), _windex(0), _buflength(0),
        _thread(osPriorityNormal, 1024, nullptr, "commandinterface"), _running(false), _response_sent(false) {
        _cbuf = (char*)malloc(CBUFSIZE);
        _unbufferedserial.attach(callback(this, &CommandInterface::callbackSerial), SerialBase::RxIrq);
    }
```

The ISR callback fills a circular buffer for each char arriving:
```
// ISR context
void CommandInterface::callbackSerial() {
    char c;
    while(_unbufferedserial.readable()) {
        _unbufferedserial.read(&c, 1);
        if (_cbuf == NULL || _buflength >= CBUFSIZE) return; // cbuf full
        _cbuf[_windex++] = c;
        // _unbufferedserial.write(&c, 1);
        ++_buflength;
        if (_windex >= CBUFSIZE) _windex = 0;
    }
}
```

The thread checks for characters in the software buffer and processes them:
```
void CommandInterface::serialProcessor() {
    std::string command;
    Timeout timeout;
    int32_t counter = -1;
    while(_running) {
        if (counter == 0) {
            WARNING("command buffer timeout");
            FINE("command size before clear: %d bytes", command.capacity());
            command.clear();
            std::string().swap(command); // release memory to avoid overflows
            //command.shrink_to_fit(); // not working
            FINE("command size after clear: %d bytes", command.capacity());
            counter = -1;
            resetSerial();
        }
        if (counter > 0) counter--;
        if (_cbuf == NULL || _buflength == 0) continue; // wait for data in cbuf
        if (command.size() < CBUFSIZE) {
            // printf("%i,", _buflength);
            printf("%c", _cbuf[_rindex]);
            command.push_back(_cbuf[_rindex++]);
            --_buflength;
            if(_rindex >= CBUFSIZE) _rindex=0;
            counter = TIMEOUT;
            
            uint16_t c_size = command.size();
            std::string s = "";
            if (c_size >= 3) {
[... 159 lines cut ...]

The complete files can be downloaded here:
[commandinterface.zip](https://github.com/ARMmbed/mbed-os/files/4962591/commandinterface.zip)
[test.zip](https://github.com/ARMmbed/mbed-os/files/4962602/test.zip)

#### Target(s) affected by this defect ?

NUCLEO_F091RC

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM  

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-v6.2.0  
https://github.com/ARMmbed/mbed-os/#a2ada74770f043aff3e61e29d164a8e78274fcd4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.4

#### How is this defect reproduced ? 

Create a small project with a similar implementation and check the reason of the issue. You can start with the files attached.

lefebvresam, 2020-07-23:
adapted

0xc0170, 2020-07-23:
Do you have any other target (different HAL implementation would be a good one) to test with ? Is this in generic implementation (drivers) or target (HAL) ? 
It is not clear to me after reading the bug report, but I might have missed it.

lefebvresam, 2020-07-23:
I don't know exactly what you mean. I use the mbed-os as discribed and the Nucleo board.

mbed_app.json:
```
{
    "macros": [
        "NDEBUG=1",
        "MBED_HEAP_STATS_ENABLED=1",
        "MBED_THREAD_STATS_ENABLED=1",
        "OS_TIMERS=0"
    ],
    "target_overrides": {
        "*": {
            "platform.memory-tracing-enabled": true,
            "platform.callback-nontrivial": true,
            "target.printf_lib": "std"
        },
        "NUCLEO_F091RC": {
            "events.shared-dispatch-from-application": true,
            "rtos.main-thread-stack-size": 3072
        }
    }
}
```

USB uart is debug port and the command interface is instantiated as:
```
_commandinterface(queue, PC_10, PC_11, 115200) --> using uart 3 or 4
```

0xc0170, 2020-07-31:
I was asking if you got another target to test with. I assume this is a bug in the target implementation rather than upper drivers layer. Have you verified what flags are being set/cleared ? Prior the transition and during the transition (looking at the MCU registers).

cc @ARMmbed/team-st-mcd for review

lefebvresam, 2020-07-31:
No, I only have one development board and I didn't went in depth analysis.

jeromecoutant, 2020-08-10:
@kjbracey-arm

0xc0170, 2020-08-13:
Kevin is away entire month, and some of us as well partly.

lefebvresam, 2020-08-13:
Ok note that for this purpose of serial interfacing since mbed 6.X there is a better alternative with BufferedSerial, but however the functionality of UnbufferedSerial needs to be perfect too and the ticket stays relevant. See https://github.com/ARMmbed/mbed-os/issues/13422

kjbracey-arm, 2020-08-17:
You don't in general get one byte per interrupt, so you do have do do the "while readable" thing in the ISR, so that's correct. Your handler looks correct to me - it's basically what `BufferedSerial` does.

The next thing is that the target HAL must not clear any interrupt latch after calling your handler. If it has to, it must do it before the handler call. This sort of error has been seen before, leading to similar symptoms.

But I don't believe the ST drivers have such a flaw, or we would have seen it in `BufferedSerial`.

I suspect the issue may be the lack of synchronisation between your foreground code and the interrupt handler. You need to be using a critical section or atomics.

This implementation could use atomics, due to the way it works - the shared state is monitored purely by `_buflength`.  Make `_buflength` be an `mstd::atomic<int>`/`mstd::atomic_int` from `<mstd_atomic>`, then I think it might start working without code changes. The breakage probably happens when an ISR hits in the middle of the `--_buflength`.

lefebvresam, 2020-08-19:
atomic<int> _buflength is not supported. I get following error:

```
Link: hmipanel
c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/NUCLEO_F091RC/GCC_ARM/source/commandinterface/commandinterface.o: in function `std::__atomic_base<int>::fetch_add(int, std::memory_order)':
c:\program files (x86)\gnu tools arm embedded\9 2019-q4-major\arm-none-eabi\include\c++\9.2.1\bits/atomic_base.h:541: undefined reference to `__atomic_fetch_add_4'        
c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/NUCLEO_F091RC/GCC_ARM/source/commandinterface/commandinterface.o: in function `std::__atomic_base<int>::fetch_sub(int, std::memory_order)':
c:\program files (x86)\gnu tools arm embedded\9 2019-q4-major\arm-none-eabi\include\c++\9.2.1\bits/atomic_base.h:551: undefined reference to `__atomic_fetch_sub_4'        
collect2.exe: error: ld returned 1 exit status
[ERROR] c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/NUCLEO_F091RC/GCC_ARM/source/commandinterface/commandinterface.o: in function `std::__atomic_base<int>::fetch_add(int, std::memory_order)':
c:\program files (x86)\gnu tools arm embedded\9 2019-q4-major\arm-none-eabi\include\c++\9.2.1\bits/atomic_base.h:541: undefined reference to `__atomic_fetch_add_4'        
c:/program files (x86)/gnu tools arm embedded/9 2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld.exe: BUILD/NUCLEO_F091RC/GCC_ARM/source/commandinterface/commandinterface.o: in function `std::__atomic_base<int>::fetch_sub(int, std::memory_order)':
c:\program files (x86)\gnu tools arm embedded\9 2019-q4-major\arm-none-eabi\include\c++\9.2.1\bits/atomic_base.h:551: undefined reference to `__atomic_fetch_sub_4'        
collect2.exe: error: ld returned 1 exit status

[mbed] ERROR: "c:\python27\python.exe" returned error.
       Code: 1
       Path: "C:\D\mbed-projects\hmipanel"
       Command: "c:\python27\python.exe -u C:\D\mbed-projects\hmipanel\mbed-os\tools\make.py -t GCC_ARM -m NUCLEO_F091RC --source . --build .\BUILD\NUCLEO_F091RC\GCC_ARM" 
       Tip: You could retry the last command with "-v" flag for verbose output
---
```

But in the foreground I did:
```
            CriticalSectionLock::enable();
            _buflength--;
            c = _cbuf[_rindex++];
            CriticalSectionLock::disable();
```

And in the constructor:
`NVIC_SetPriority(USART3_8_IRQn, 255);`

But I still got a data loss when bunches of serial data were coming in frequently during SPI-communication. It resulted in interpretation errors. This was the case when I touched my display and the controller was communicating to the GPU over SPI to refresh the image. After investigation I reduced the serial speed from 350000Bd to 120000Bd and this invoked much more stability. There was no data loss anymore. Maybe when there is SPI-communication going on, the interrupts are not handled in the right priority and when a lot of serial data is coming in, you can have hardware buffer overflows at high speeds. Is there any suggestion to improve that balance?

kjbracey-arm, 2020-08-19:
That needs to be **`mstd::atomic`** from Mbed OS's  **`<mstd_atomic>`**, not `std::atomic` from the toolchain's `<atomic>`. The Mbed `mstd`-namespaced version isolates you from variance and implementation limitations in the toolchains, and provides atomic-against-IRQ guarantees, which the standard C libraries don't necessarily do.

Critical section lock would be equivalent to the atomic if it was an M0(+) - the atomic is better on other devices. You don't in this case need a lock around the buffer-and-rindex manipulation, as rindex isn't touched by the interrupt, and the buffer accesses are synchronised by the buflength protection.

Did that solve the "stuck character" problem?

Priority 255 is minimum - that will have made UART reliability worse. Undo that. By default all interrupts are priority 0 (maximum), so that will make it standard priority again. To make serial interrupts higher priority you'd have to go through and set all the *others* lower.  But interrupt priorities don't help if the IRQ latency is caused by people holding critical section locks for a long time. 

SPI shouldn't particularly upset the serial, unless you're using it from an interrupt context, which you shouldn't. If you are, then at least make that interrupt handler be lower priority. IEEE 802.15.4 radio drivers used to use SPI from interrupt, but moved to using the high priority event queue+thread instead to avoid that problem.

But generally, on an M0 device like yours I'd be a bit doubtful about >115200 serial being reliable with PIO. As far as I recall, the device has no serial FIFO, so you have a pretty tight IRQ latency requirement. 80us at 115200, and proportionately lower as you get faster. That's hard to meet in a conventional OS+IRQ setup.

If you could make the UART interrupt be higher priority than "normal" and make "enter critical" only mask "normal" priority interrupts, you'd be sorted, but M0 lacks the hardware ability to selectively mask levels. We have some platforms which need fast interrupt response time, eg for BLE, that work like that, but they require M3/M4, which have the BASEPRI register.

You might be able to make it work with asynchronous serial reads, which can use DMA (do they on STM?), but they introduce their own problems, and might not fit your data protocol. The API forces you to have fixed-length transactions to set up the DMA.

jeromecoutant, 2020-09-09:
@lefebvresam any update ?

lefebvresam, 2020-09-09:
No currently I have a working version with the tweaks I did and my focus is on the POC. I saved your comments in my code in order to do further improvement as soon as I re-use the code or I want to extend the functionality of my product.

jeromecoutant, 2021-02-23:
@0xc0170 @lefebvresam Let's close this issue ?

lefebvresam, 2021-02-23:
When solved ok

jeromecoutant, 2021-02-24:
> I have a working version with the tweaks I did and my focus is on the POC.

Maybe you can start sharing them in a patch ?

lefebvresam, 2021-02-24:
> > I have a working version with the tweaks I did and my focus is on the POC.
> 
> Maybe you can start sharing them in a patch ?

Yes I can do but not before April 4th because I'm currently working on another project.

lefebvresam, 2021-04-16:
The more stable implementation I use:
[commandinterface.zip](https://github.com/ARMmbed/mbed-os/files/6324198/commandinterface.zip)

MubeenHCLite, 2021-06-25:
I do not have the exact platform, but i have tried your code on a couple of platforms. Unable to reproduce the issue.
Many enhancements went in after the issue was raised. 
Please re-check with the current mbed release and close the issue if there are no problems at your side.

MubeenHCLite, 2021-08-03:
@lefebvresam 
Reminder to close the issue.

lefebvresam, 2021-08-03:
Project is currently out of (business) scope. When it will be picked up again I can re-check the issue. Is there any way to archive this thread in order to re-activate when necessary?

prasannaven, 2021-08-03:
@0xc0170 can you suggest

MubeenHCLite, 2021-08-06:
@0xc0170 
A gentle reminder, please let know if there is a way to archive this issue.

0xc0170, 2021-08-06:
I'll close it, reopen with an update

closed 2021-08-06 by 0xc0170