ARMmbed/mbed-os #13343: version confusion

tdjastrzebski, 2020-07-23:
It seems using `mbed new .` command I am not getting the latest release 6.2 or at least not all the files are the latest.

In particular I am looking at the [cmsis_nvic.h](https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32F7/TARGET_STM32F769xI/cmsis_nvic.h) file.
In the version available online the incorrect table length issue #12793 appears to be fixed, `NVIC_NUM_VECTORS = 126`.
Fix was merged on April 28th.

However, version obtained using `mbed new .` does not contain this fix (NVIC_NUM_VECTORS=114), the file was last modified 4 yrs ago but the last commit I see looking at Git history is correctly dated Jul 23.

I am puzzled.

Btw, you may want to update [README.md](https://github.com/ARMmbed/mbed-os/blob/master/README.md) file. It still refers **mbed-os-5-docs repository**. It also would not hurt to update this file every time release is made or add some other file indicating product version.

0xc0170, 2020-07-27:
> Btw, you may want to update README.md file. It still refers mbed-os-5-docs repository. It also would not hurt to update this file every time release is made or add some other file indicating product version. E.g. Version.txt

The name of the repo is confusing - that is what you get when you use version number without expecting a major version to be released 🙄  

> version confusion

Please provide more detailed title for an issue, is this issue related to 5.15.x or 6.x version issue? 

What version do you get? or SHA if you can get that one from the repository? From the issue referenced (#12793), it should be part of 6.0.0. Was it overwritten post 6.0.0 or the version you get is not 6.x but rather 5.15.x ?

tdjastrzebski, 2020-07-27:
@0xc0170 I will be happy to provide more details but first please tell me how do I know what version `mbed new .` downloads?

0xc0170, 2020-07-31:
if you browse into mbed-os, you can open mbed.h file to check, or use git log to get sha/tags.

tdjastrzebski, 2020-08-02:
@0xc0170 Thanks Martin, I am showing it is actually https://github.com/ARMmbed/mbed-os/blob/master/platform/mbed_version.h header which contains the version. The point I was trying to make was that, unless you knew exactly where to look, it was hard to tell which version folder contains. Typically, software releases contain some release notes file with version and all relevant info so one who opens his/her archive yeas later knows which version it was. I think it makes perfect sense for mbed-os to follow this standard practice.

Now I can tell that executing `mbed new .` I get version **5.15.4** which is probably why the content differs from what I expected and the issue #12793 is still present.

jeromecoutant, 2020-08-10:
Hi

> Now I can tell that executing `mbed new .` I get version **5.15.4** which may indicate a problem and which is why the content differs from what I expected (latest version 6.2.0) and the issue #12793 is still present.

Yes, #12793 is only corrected on mbed-os-6 branch

tdjastrzebski, 2020-08-10:
Hi, but still, I think the fact that executing `mbed new .` one will get version **5.x** is not well documented.
I am afraid most users expect the latest released version.

0xc0170, 2020-08-10:
executing `mbed new .` gets you the latest release. As there are currently two different versions being released often (Mbed Os 5.15.x and Mbed OS 6.), there is a clash between these two. This `latest` was always problematic and having multiple different releases (lets consider LTS here), this does not scale.

Adding mbed-os-tools here to be aware of another why not use `latest` as we do.

cc @ARMmbed/mbed-os-tools

0xc0170, 2020-08-10:
> Another issue is that there is no obvious way to tell which version is being obtained. There is no standard README or RELEASE NOTES file containing version info in the mbed-os root folder.

👍 I'll create an issue for this so @ARMmbed/mbed-os-maintainers with docs team can review how to improve the releases

0xc0170, 2020-08-10:
I created internal  ticket https://jira.arm.com/browse/IOTMAINTAIN-193 for maintainers

adbridge, 2020-11-17:
@tdjastrzebski The mbed-os version is embedded in the code itself. Please take a look at 
https://github.com/ARMmbed/mbed-os/blob/master/platform/include/platform/mbed_version.h 
This file is updated for every release.

tdjastrzebski, 2020-11-18:
@adbridge That is my point. Version info is well hidden, while, typically, it is available somewhere in the product root folder, e.g. in README or a similar file.

adbridge, 2020-11-19:
@tdjastrzebski  we'll raise it with our management team as an improvement suggestion :)