ARMmbed/mbed-os #13350: [STM32] Add USBDEVICE support to more targets

DesktopMan, 2020-07-24:
### Description of defect

Would be very nice if more STM targets supported USBDEVICE.

#### Target(s) affected by this defect ?

- MCU_STM32F103x8 
- MCU_STM32F103xB
- NUCLEO_F103RB

Allmoz, 2020-07-25:
To enabling that into STM32F1 you need USBPhyHw.h to define:

#define USB_IRQn USB_LP_IRQn

so line 39 that is:

#if defined(TARGET_STM32F3) || defined(TARGET_STM32WB)

could be:

#if defined(TARGET_STM32F1) || defined(TARGET_STM32F3) || defined(TARGET_STM32WB)

(if you wanna do it only on a custom target, append it to PeripheralNames,h. I did it this)

Then you have to define in your PeripheralNames,h: 

typedef enum {
    USB_FS = (int)USB_BASE,
} USBName;

and you have to define in your PeripheralPins,h: 

MBED_WEAK const PinMap PinMap_USB_FS[] = {
    {PA_11,     USB_FS, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0)}, // USB_DM
    {PA_12,     USB_FS, STM_PIN_DATA(STM_MODE_INPUT, GPIO_NOPULL, 0)}, // USB_DP
    {NC, NC, 0}
};

in PinNames.h:

USB_DM = PA_11,
USB_DP = PA_12,

and add to your custom_targets.json :

"device_has_add": ["USBDEVICE"],

0xc0170, 2020-07-27:
@Allmoz Thanks for the details!

@DesktopMan As this is feature request, it should be in forums.mbed.com. I'll close this as non technical issue. 
Based on the details above, you should be able to add and test more targets with USB functionality.

closed 2020-07-27 by 0xc0170

jeromecoutant, 2020-08-10:
@Allmoz 

Many thx for the comment!
Do you think you could propose a pull request ?

Allmoz, 2020-08-10:
Ok

Allmoz, 2020-08-10:
Done