ARMmbed/mbed-os #13352: PSoC6x KVstore failures

maclobdell, 2020-07-24:
### Description of defect

When testing PSoC62 and PSoC64 targets with Pelion Client, we have noticed an intermittent issue when storing key/value pairs in the KVStore in internal memory. This results in Pelion Client error messages such as corrupted or malformed values retrieved from the storage. This occurs even after full erase of the flash before re-writing values. 

Mysteriously these issue can go away just by adding a printf here or there around where the writes are happening. 

We suspect that this problem could be caused by an alignment issue with the flash driver. 

#### Target(s) affected by this defect ?

CY8CPROTO_062_4343W
CY8CKIT_064B0S2_4343W

#### Toolchain(s) (name and version) displaying this defect ?

Arm compiler 6

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.1.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

1.10.2

#### How is this defect reproduced ? 

import mbed-os-example-pelion
build for the selected target 
most of the time everything works fine, but on rare attempts, there is corrupted values in the kvstore

0xc0170, 2020-07-27:
cc @ARMmbed/team-cypress

LDong-Arm, 2020-11-25:
> We suspect that this problem could be caused by an alignment issue with the flash driver.

Hi @maclobdell, were you using a KVStore on the QSPI/SPI flash? In that case would you like to see if #13947 fixes the issue?