ARMmbed/mbed-os #13354: Stream.h

sstaub, 2020-07-25:
### Description of defect

#### Target(s) affected by this defect ?

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2020-07-27:
I can't find a pull request that would remove `Stream.h` , was it included previously ?

My assumption would be this removed it via classes that depend on `Stream.h`: https://github.com/ARMmbed/mbed-os/pull/12410 . It was a clean up of deprecated functionality.

I can see USB drivers are using `Stream.h`.

@kjbracey-arm @rajkan01 do you remember any detail about Stream class

rajkan01, 2020-07-27:
> I can't find a pull request that would remove Stream.h , was it included previously ?
> My assumption would be this removed it via classes that depend on Stream.h: #12410 . It was a clean up of [...]

Yes, previously `Stream.h` header included via `drivers/Serial.h` header file. As part of the cleanup, the deprecated class was removed and stream header in #12410.
I am not sure why that header file was not included directly in `mbed.h` header file instead

0xc0170, 2020-07-27:
> I am not sure why that header file was not included directly in mbed.h header file instead

A missed bit. If the class is used (it is in USB drivers) so should be exposed, shouldn't be?

0xc0170, 2020-09-16:
@rajkan01 worth fixing ? We can send a pull request

cc @adbridge if you got a chance to review this and fix?

0xc0170, 2021-01-26:
@adbridge Can you look at this one?

adbridge, 2021-01-28:
@rajkan01 So if stream.h was removed because previously it was in Serial.h , shouldn't stream.h instead be included in the replacement for Serial.h ? ie bufferedserial.h and/or unbufferedserial.h ?

adbridge, 2021-01-28:
@sstaub while we debate if / where Stream.h should sit, one point to note is that the TextLCD you are using is actually from Mbed 2 and will have not been tested / ported to Mbed 5 or Mbed 6 so there are no guarantees of it working...

sstaub, 2021-01-28:
I have made some LCD libraries which are updated for Mbed 6 I published also on my Mbed site, I included the Stream.h manually, but I think this class should included in mbed.h

0xc0170, 2021-01-28:
Thanks for the feedback, it looks to me it was previously exposed as @rajkan01 wrote and it's public API:

```
#ifndef MBED_STREAM_H
#define MBED_STREAM_H

#include "platform/platform.h"
#include "platform/FileLike.h"
#include "platform/NonCopyable.h"
#include "platform/mbed_toolchain.h"
#include <cstdio>
#include <cstdarg>

namespace mbed {
/** \addtogroup platform-public-api */
/** @{*/

```

kjbracey-arm, 2021-01-28:
I've been pushing for a long time for `Stream` to be deprecated. It's a helper class that helps you implement the `FileHandle` API, but it does so very badly. I'd much rather people implemented `FileHandle` directly than use `Stream` to do it.

Here was my attempt to do that deprecation: https://github.com/ARMmbed/mbed-os/pull/5655

I didn't manage to win the debate then.

We've since replaced `Serial`, but `Stream` remains. And it's no better than it was then.

Previously it would have been picked up accidentally by mbed.h getting it from Serial.h. As it is still public API, mbed.h should be including it explicitly now for anyone using it.

I'd still like to deprecate it though.

sstaub, 2021-01-28:
Sorry, this absolut ugly. Does anybody from ARM read the forum? Most people are not happy about the way Mbed goes.
How should we implement an easy printf functionality for libraries? You APIs get more and more stupid.

closed 2021-02-03 by 0xc0170