ARMmbed/mbed-os #13360: ThisThread::flags_wait_any_for has wrong flags in implementation

caoyuan96421, 2020-07-28:
### Description of defect
The method `ThisThread::flags_wait_any_for` is using `osFlagsWaitAll` instead of `osFlagsWaitAny`. See below snippet of code in `ThisThread.cpp` line 201.

```C++
uint32_t ThisThread::flags_wait_any_for(uint32_t flags, Clock::duration_u32 rel_time, bool clear)
{
    return flags_wait_for(flags, rel_time, clear, osFlagsWaitAll);
}
```

#### Target(s) affected by this defect ?
All

#### Toolchain(s) (name and version) displaying this defect ?
Doesn't matter

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.1.0 a6207ca

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Doesn't matter

#### How is this defect reproduced ? 
Obvious from code

0xc0170, 2020-07-29:
Thanks for reporting. Can you send a pull request fixing it?

caoyuan96421, 2020-07-30:
> Thanks for reporting. Can you send a pull request fixing it?

See PR #13374

PR #13374 (2020-08-10): Fixed ThisThread flag_wait_any functions (fix #13360)

### Summary of changes 

Fixed ThisThread flag_wait_any functions
See #13360.

#### Impact of changes 

None

#### Migration actions required
commit e127bb24 (2020-08-10): Merge pull request #13374 from caoyuan96421/bugfix2

Fixed ThisThread flag_wait_any functions (fix #13360)
closed 2020-08-10 by 0xc0170