ARMmbed/mbed-os #13368: Cordio BLE: GAP::cancelConnect() results in Hard Fault/memory corruption

mg-unleashed, 2020-07-29:
### Description of defect

#### Target(s) affected by this defect ?
NRF52

#### Toolchain(s) (name and version) displaying this defect ?
GCC9.2.1 (GNU Tools 9-2019-q4-major)

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-5.15.1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
Mbed-cli 1.10.4

#### How is this defect reproduced ? 
1. Call Gap::connect()
2. Call Gap::cancelConnect() before a connection is established.

Detailed call stack::
DmConnClose() at dm_conn.c:1,027 0x197cc	
ble::pal::vendor::cordio::Gap<ble::generic::GenericGap<ble::pal::vendor::cordio::Gap, ble::pal::SecurityManager<ble::pal::vendor::cordio::CordioSecurityManager<ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> > >::cancel_connection_creation_() at CordioPalGap.tpp:278 0x1221e	
ble::pal::Gap<ble::pal::vendor::cordio::Gap<ble::generic::GenericGap<ble::pal::vendor::cordio::Gap, ble::pal::SecurityManager<ble::pal::vendor::cordio::CordioSecurityManager<ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> > >, ble::generic::GenericGap<ble::pal::vendor::cordio::Gap, ble::pal::SecurityManager<ble::pal::vendor::cordio::CordioSecurityManager<ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> > >::cancel_connection_creation() at PalGap.h:1,598 0x13ba4	
ble::generic::GenericGap<ble::pal::vendor::cordio::Gap, ble::pal::SecurityManager<ble::pal::vendor::cordio::CordioSecurityManager<ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >::cancelConnect_() at GenericGap.tpp:828 0x13ba4	
ble::interface::Gap<ble::generic::GenericGap<ble::pal::vendor::cordio::Gap, ble::pal::SecurityManager<ble::pal::vendor::cordio::CordioSecurityManager<ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> >, ble::generic::GenericSecurityManager<ble::pal::vendor::cordio::CordioSecurityManager, ble::vendor::cordio::SigningEventMonitor> > >::cancelConnect() at Gap.tpp:311 0x13bac	

Inside DmConnClose a message is created with the following header info:
event = DM_CONN_MSG_API_CLOSE
param = DM_CONN_ID_NONE (0). 
status = 0
clientId = DM_CLIENT_ID_APP

The assignment of param = DM_CONN_ID_NONE is the source of the error when the message is processed.

Then the message is ultimately processed by dm_conn.c:dmConnMsgHandler() pMsg->param is used to determine the ccb, however DM_CONN_ID_NONE is incorrectly processed by dmConnCcbById() and an invalid ccb pointer is returned. This results in dmConnSmExecute() operating on an invalid ccb pointer. 

In my case, it generates a Hard Fault, but depending on the memory layout, memory corruption could also result.

Looking at this issue from a more systematic view, it seems cancelConnect() needs a connection identifier to know which connection to cancell, however this would require connect() to provide such an identifier. 

Proposed Resolution:
Modify cancelConnect() to require the peer address of the connection to be cancelled. This would allow Gap<EventHandler>::cancel_connection_creation_)_ to look up the connId with the help of dmConnCcbByBdAddr().

0xc0170, 2020-07-31:
cc @mbed-os-pan Please review

paul-szczepanek-arm, 2020-08-03:
Thank you for the detailed report and analysis. I will try to schedule this ASAP.

paul-szczepanek-arm, 2020-08-03:
tracked https://jira.arm.com/browse/IOTPAN-651

paul-szczepanek-arm, 2020-08-11:
Thank, this is now fixed: https://github.com/ARMmbed/mbed-os/pull/13390.

closed 2020-08-13 by 0xc0170