ARMmbed/mbed-os #13372: ScanParameters class uses incorrect types for interval & window members

mg-unleashed, 2020-07-29:
### Description of defect
ScanParameter class members interval & window use the wrong type.

ScanParameter Member definitions are
scan_window_t interval;
scan_interval_t window;

and should be:
scan_interval_t interval;
scan_window_t window;

#### Target(s) affected by this defect ?
N/A

#### Toolchain(s) (name and version) displaying this defect ?
N/A

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-5-15-1

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
N/A

#### How is this defect reproduced ? 
N/A

0xc0170, 2020-07-31:
@ARMmbed/mbed-os-pan Please review

paul-szczepanek-arm, 2020-07-31:
Thank you for spotting this and letting us know. The unit names are indeed swapped. The underlying unit for both is 625us so whilst internally the unit names are incorrect from the functionality point of view there is no difference. This will be patched soon on master.

mg-unleashed, 2020-07-31:
I agree, since the underlying types are equivalent there is not functional difference, but it is misleading and confusing. Thanks

paul-szczepanek-arm, 2020-08-13:
this is now fixed on master

closed 2020-09-24 by pan-