ARMmbed/mbed-os #13376: Custom board with external TRNG

boraozgen, 2020-07-30:
### Description of defect

I am trying to use an external cryptoprocessor with TRNG function to add TRNG capability to a MCU on a custom board (STM32F411xE in this case). Adding `TRNG` to `device_has_add` in `custom_targets.json` requires me to implement the `trng_api.h` functions and the `trng_s` data structure as described [here](https://os.mbed.com/docs/mbed-os/v6.2/porting/entropy-sources.html). Although it possible to simply add the `trng_api` implementation source to my target directory, I cannot define the `trng_s` data structure as it is normally defined in `objects.h`, which is included from the `TARGET_STM32F411xE` directory. I could define the structure in `PinNames.h` or `PeripheralNames.h`, but this would not be appropriate to their logical purposes. There should be a better way to define the TRNG structure.

#### Target(s) affected by this defect ?

Custom board using STM32F411xE

#### Toolchain(s) (name and version) displaying this defect ?

Not relevant.

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not relevant.

#### How is this defect reproduced ? 

Porting issue, not a defect.

boraozgen, 2020-07-30:
Now that I am trying out the aformentioned workaround of putting the structure in the other header files, I realized that the `trng_api.c` from `TARGET_STM` is trying to be compiled and of course fails due to the missing RNG HAL. I guess I have to implement `mbedtls_hardware_poll` directly without using the Mbed OS API. This means adding `MBEDTLS_ENTROPY_HARDWARE_ALT` macro to the target definition manually. I think it would be more elegant if it was possible using the `device_has_add` property.

0xc0170, 2020-07-30:
> Now that I am trying out the aformentioned workaround of putting the structure in the other header files, I realized that the trng_api.c from TARGET_STM is trying to be compiled and of course fails [...]

I believe you found a solution. The `device_has` currently is as it is, enabling what HAL provides. In this case, you want to add an external module, device_has was not design to be that way thus `object.h` already defined by HAL. There is no drivers API for trng as the only consumer is meant to be mbedtls. Thus using its functionality to overwrite the default trng module and using an external one is a way to go.

boraozgen, 2020-07-30:
Alright, I can live with that. Thanks for the support!

closed 2020-07-30 by boraozgen

jeromecoutant, 2020-08-13:
Hi
Quick comment:
Many STM32F4 support RNG feature, but not STM32F411xx.

boraozgen, 2020-08-13:
As mentioned in the issue, this is a custom board with HW crypto chip with TRNG capability.