ARMmbed/mbed-os #13388: AsynchSerial: No public way to access read/write

AGlass0fMilk, 2020-08-01:
### Description of defect

Recent updates involving the `Serial` implementations in Mbed-OS have made it impossible to access AsynchSerial methods `SerialBase::read` and `SerialBase::write`. It appears the two replacement subclasses, `UnbufferedSerial` and `BufferedSerial`, do not make use of these transfer methods and instead rely on single-byte transfers.

I am working on a serial bootloader operating at 115200 baud and have found it impossible to use the new public subclasses to receive in-tact data. I believe the cause is that the new `Unbuffered/BufferedSerial` subclasses operate in a manner that is too slow to capture all data properly...

My target is an nRF52840.

Could be related to #13339 

#### Target(s) affected by this defect ?
nRF52840, possibly all targets

#### Toolchain(s) (name and version) displaying this defect ?
GCC ARM:
`arm-none-eabi-gcc --version`:
```
arm-none-eabi-gcc (GNU Tools for Arm Embedded Processors 9-2019-q4-major) 9.2.1 20191025 (release) [ARM/arm-9-branch revision 277599]
```

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.1.0 (a6207cadad0acd1876f436dc6baeddf46c42af06)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli version: 1.10.4

#### How is this defect reproduced ? 

Try to receive more than one byte of serial data at a time...

0xc0170, 2020-08-13:
cc @ARMmbed/mbed-os-core

loverdeg, 2020-08-15:
@kjbracey-arm

kjbracey-arm, 2020-08-17:
The HAL async read calls are a horrible fit to the `FileHandle` read calls (as they read exact length, rather than "whatever is available up to buffer size"), so they don't mesh well. That's been one of the things on a "to do" list for HAL API update for ages.

`BufferedSerial` could use the async write call now to implement `FileHandle::write`, but that's never been done largely for backwards compatibility - fear of misbehaving little-used async APIs on some targets. Maybe could be a config option. Probably not a good idea for `UnbufferedSerial`, as that is expecting to see use from IRQ context. Although I guess it could get the same "synchronous if in IRQ" check that `UnbufferedSerial` got.

For people that really do want to use the async stuff manually, I guess the unification on `FileHandle` has got in the way. There was a desire to unify `RawSerial` with the internal `DirectSerial` from mbed_retarget.cpp, and the resulting `UnbufferedSerial` has given you the `FileHandle` of the latter, but it's lost you full access to the `SerialBase`. :(

I suppose there's no reason not to pass-through the asynchronous APIs in `UnbufferedSerial`. I kind of forget the design decision discussions, but `SerialBase` being private for `BufferedSerial`/`UARTSerial` was very deliberate, as most use of underlying `SerialBase` APIs would break the IRQ handler + buffer logic; async APIs definitely would.

It's somewhat less obvious why it should be private for `UnbufferedSerial`. But just making it `public` wouldn't be sufficient anyway, as we have to deal with  `read`/`write` being shadowed - all overloads need to be in the derived class, IIRC.

So I'd say, yes, the asynch stuff should probably be exposed in `UnbufferedSerial`, but not sure whether to make `SerialBase` a public base and add extra passthrough `read`/`write` overloads, or to keep it private with selective passthrough.

loverdeg, 2020-08-18:
Another one of my gut feels but file handles have always felt wrong to me.   We have probably had this exchange of comments before.

robmosys, 2021-02-25:
Would really love to see this fixed, will be patching up locally for the moment which will probably break when we upgrade to a version where async is fixed :-) 
If you want, can add in-field report of how un-buffered serial behaves with these public.
Or... maybe submit a patch.