ARMmbed/mbed-os #13389: nRF: serial_api boolean flag usage is inconsistent

AGlass0fMilk, 2020-08-01:
### Description of defect

There are several flags in the control block of the nRF `serial_api.c` implementation for asynchronous serial. The specific ones involved in this issue are:

- `obj->serial.rx_asynch`
- `obj->serial.tx_asynch`
- `obj->serial.rx_in_progress`
- `obj->serial.tx_in_progress`

I'm not sure how this hasn't come up before, but I vividly remember using the asynchronous serial API on nRF targets in the past.

This whole investigation started when I tried to use Mbed's serial to receive data at a decent rate (115200 baud). See #13388. The problem I saw was that my python script was sending 128 bytes over serial but my Mbed target was only receiving a random number between 1 and ~50 bytes, so it would just hang in a waiting loop. I believed the cause to be that the synchronous serial implementations (essentially just reading/writing bytes 1 by 1) was too slow and would drop bytes in between.

So I started working with the asynchronous serial API (by making a new class that made these methods public). Now I have a new problem! My Mbed program waits to receive a 0-byte from the python script at the beginning and then writes a `1` to the serial port to tell the python script it's there. The python script then stops and waits to receive a `2` before starting to write out chunks of a binary file (this is a serial bootloader implementation).

See the code below:

```
    // Set up the serial interface
//    mbed::BufferedSerial pc(STDIO_UART_TX, STDIO_UART_RX, MBED_CONF_APP_SERIAL_BOOTLOADER_BAUD);
    RawSerial pc(P0_3, P0_4, MBED_CONF_APP_SERIAL_BOOTLOADER_BAUD);
    pc.format(8, mbed::SerialBase::None, 1);

    // Wait for a specified timeout until just booting
    rtos::Kernel::Clock::duration time_waited = 0ms;
    bool timed_out = true;
    uint8_t c = 1;
    while(time_waited < BOOT_WAIT_TIMEOUT) {
        // The serial bootloader should send 0-bytes until we respond we're ready, look for that
        if(rx_complete) {
            rx_complete = false;
            if(c == 0) {
                timed_out = false;
                break;
            }
        }

        pc.read(&c, 1, mbed::callback(rx_handler));

        // Otherwise, keep waiting
        rtos::ThisThread::sleep_for(100ms);
        time_waited += 100ms;
    }

    // Serial bootloader timed out, boot the main application
    if(timed_out) {
        return;
    }

    // Serial bootloader connected, send an acknowledgement
    c = 1;
    pc.write(&c, 1, mbed::callback(tx_handler));
    while(!tx_complete) { }
    tx_complete = false;

    // Queue up the read
    uint8_t buf[128] = { 0 };
    pc.read(buf, 64, mbed::callback(rx_handler));

    c = 2;
    pc.write(&c, 1, mbed::callback(tx_handler));
    while(!tx_complete) { }
    tx_complete = false;

    while(!rx_complete) { }
    rx_complete = false;

    while(true) { }
```
I found that when the first `read` callback occurs, the following handler is executed:

https://github.com/ARMmbed/mbed-os/blob/e57e325274851216c7c3192d25c8a52672c1106a/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/serial_api.c#L490-L535

Notice line 522, it is checking a flag that would be cleared above so that portion will always be true.

In the first flag handler (`NRF_UARTE_EVENT_ENDRX`) we will call `nordic_nrf5_uart_event_handler_endrx_asynch`:

https://github.com/ARMmbed/mbed-os/blob/e57e325274851216c7c3192d25c8a52672c1106a/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/serial_api.c#L463-L487

...ookay... Why in line 471 is the `rx_asynch` flag being set to false? We're not exiting asynch mode, the asynch RX operation is just complete. The next line makes total sense, flagging `rx_in_progress` to false.

For some reason whenever I get a `NRF_UARTE_EVENT_ENDRX` event I also get a `NRF_UARTE_EVENT_RXSTARTED` (possibly misconfigured event short somewhere?), which then executes `nordic_nrf5_uart_event_handler_rxstarted`:

https://github.com/ARMmbed/mbed-os/blob/e57e325274851216c7c3192d25c8a52672c1106a/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/serial_api.c#L438-L460

I don't think this handler makes any sense in the asynch use case. There should **NOT** be a 1-byte long buffer queue'd up when the asynch buffers are provided by the application. Moving on...

Now the main application writes the `1` byte out. During that call to `SerialBase::write` eventually `nordic_nrf5_serial_configure` is called to configure the UART for TX. Since line 471 set `rx_asynch` (ie: the current mode of the UART) to `false`, the following if statement is now executed:

https://github.com/ARMmbed/mbed-os/blob/e57e325274851216c7c3192d25c8a52672c1106a/targets/TARGET_NORDIC/TARGET_NRF5x/TARGET_NRF52/serial_api.c#L796-L798

Which reconfigures the UART in asynch RX mode... alright, I guess that's okay... until...

Alright, so now we have received the byte, we have written the `1` byte response out, the main application then calls `SerialBase::read`:

https://github.com/ARMmbed/mbed-os/blob/e57e325274851216c7c3192d25c8a52672c1106a/drivers/source/SerialBase.cpp#L425-L436

The call to `serial_rx_active` is ambiguous in terms of what it is supposed to do. The worst part is that since the previous call to `write` configured the UART to asynch RX mode (thus setting `rx_asynch` to `true`), the read operation is never started. This is because `serial_rx_active`, instead of checking `rx_in_progress` as I think it should, checks `rx_asynch`, which is now `true`...

So there seems to be a lot of inconsistency in this driver and in the asych serial API to begin with. I'm willing to help do some bug fixing but I'd like to know what state the asynch serial API is in? Is it due for a refactor? Right now it's not even public, really.

I swear I used to be able to do something so simple as receive multiple bytes over serial pretty easily in Mbed, I don't know what has changed!

#### Target(s) affected by this defect ?

nRF52840, nRF52832

#### Toolchain(s) (name and version) displaying this defect ?

All

#### What version of Mbed-os are you using (tag or sha) ?

master (e57e325274851216c7c3192d25c8a52672c1106a)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.4

#### How is this defect reproduced ? 

See above description with example code.

@hugueskamba @marcuschangarm

0xc0170, 2020-08-31:
I updated the sha, it should be fixed now.

> So there seems to be a lot of inconsistency in this driver and in the asych serial API to begin with. I'm willing to help do some bug fixing but I'd like to know what state the asynch serial API is in? Is it due for a refactor? Right now it's not even public, really.

@ARMmbed/mbed-os-hal would you be able to review?

MarceloSalazar, 2020-09-02:
@AGlass0fMilk 

> So there seems to be a lot of inconsistency in this driver and in the asych serial API to begin with. I'm willing to help do some bug fixing but I'd like to know what state the asynch serial API is in? Is it due for a refactor? Right now it's not even public, really.

I assume you're referring to the HAL API, not the Mbed Drivers API. Is this correct?
We're open to make changes at some point in the near future and would like to hear possible enhancements to be considered. Thanks.

AGlass0fMilk, 2020-10-13:
> @AGlass0fMilk
>
>
> I assume you're referring to the HAL API, not the Mbed Drivers API. Is this correct?
> We're open to make changes at some point in the near future and would like to hear possible [...]

@MarceloSalazar Sorry I hadn't seen this!

I would love to be part of the discussion when you guys are thinking of revamping it. I'll think about changes but I think a live brainstorming session would help us identify existing problems and what the next generation API could look like.

closed 2021-10-20 by ciarmcom