ARMmbed/mbed-os #13395: CellularNetwork instance not accessible

boraozgen, 2020-08-06:
### Description of defect

I am using an external cellular modem (Quectel BG96/95) and `NetworkInterface` API to connect to the internet. I would like to use the cellular registration information for diagnosis purposes in my application. I noticed that these information are conveniently read and saved in the `CellularNetwork` class, which is used by the `CellularContext` class, which however does not expose the used instance. I wonder if this is a design decision or was just overlooked in the API. Shouldn't the `CellularNetwork` instance be exposed to the application? I think the registration information, such as signal quality and cell information could be useful to any application using cellular connections.

#### Target(s) affected by this defect ?

Custom board.

#### Toolchain(s) (name and version) displaying this defect ?

API issue, not relevant.

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

API issue, not relevant.

#### How is this defect reproduced ? 

API issue, not relevant.

kivaisan, 2020-08-10:
@boraozgen You can get handle to `CellularNetwork` via `CellularContext` and `CellularDevice`:

For example:
```
NetworkInterface *iface = NetworkInterface::get_default_instance(); // "target.network-default-interface-type": "CELLULAR"
CellularContext *ctx = (CellularContext *)iface;
CellularDevice *dev = ctx->get_device();
CellularNetwork *nw = dev->open_network();
```

boraozgen, 2020-08-10:
@kivaisan open_network() creates a new AT_CellularNetwork, not a good idea if I am already connected.

boraozgen, 2020-08-10:
I see, the name of the method is confusing though. I guess it does not create a new network instance if it already exists?

kivaisan, 2020-08-10:
Yes, it will create instance only if not created already earlier. This is a public API function so name is frozen.

boraozgen, 2020-08-10:
This solution works for me, closing the issue. Nevertheless I still think the API is unclear and something like `get_network()` would be much better. Thanks for the support.

closed 2020-08-10 by boraozgen