ARMmbed/mbed-os #13398: Memory leak in nsapi_dns

boraozgen, 2020-08-07:
### Description of defect

While I was analyzing the heap stats of my application, where I connect and disconnect to a server periodically, I noticed a 17 byte memory leak occuring around every minute. When I compared the traces, I found that the address member of the dns_cache struct is allocated here:

https://github.com/ARMmbed/mbed-os/blob/c85b76ad0589d2953a4c9272b0350d1fcbe9ab96/connectivity/netsocket/source/nsapi_dns.cpp#L372

but not freed on expiry:

https://github.com/ARMmbed/mbed-os/blob/c85b76ad0589d2953a4c9272b0350d1fcbe9ab96/connectivity/netsocket/source/nsapi_dns.cpp#L399-L402

#### Target(s) affected by this defect ?

Higher layer issue, not target specific.

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.2.0
mbed-os-99.99.99 (c85b76ad0589d2953a4c9272b0350d1fcbe9ab96)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Not relevant.

#### How is this defect reproduced ? 

Connect to a TCP socket multiple times for more than a couple minutes.

#### Patch proposal:

https://github.com/boraozgen/mbed-os/commit/a699ff34068214a9e829c584401769c2ee22a319

I can open a PR if requested.

0xc0170, 2020-08-10:
> I can open a PR if requested.

@boraozgen Please create a PR, thank you.

0xc0170, 2020-08-10:
cc @ARMmbed/mbed-os-ipcore

closed 2020-08-11 by boraozgen