ARMmbed/mbed-os #13402: printf() crashes Mbed OS when passed a extern string as argument for %s format

gmgallo, 2020-08-09:
### Description of defect
It only happens when a string argument or the formatting string itself resides in another module and are declared like this: 
extern char *my_string;

Other external arguments like ints, floats, even pointers to doubles, etc. work fine. 
The problem is also present in sprintf() formatting to a buffer much larger than the extern string. 

#### Target(s) affected by this defect ?
I'm working with CY8CPROTO-062-4343W PSoC 6 development board

#### Toolchain(s) (name and version) displaying this defect ?
Mbed Studio 1.1.0

#### What version of Mbed-os are you using (tag or sha) ?

6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
The compiler and linker installed with Mbed Studio

#### How is this defect reproduced ? 
Start with the blinky sample.
1) In main.cpp  add:

void my_print();
char my_string[] = "This is my string";

call my_print() from inside main()

2) in blinkThread.cpp add:

extern char *my_string;

void my_print()
{
  printf("%s",my_string);  // Mbed OS will crash when it reaches here
}

0xc0170, 2020-08-10:
Is this an issue for Mbed OS? Have you tested your program in any other environment, for instance C++ online compiler?

`extern char my_string[];` in blinkThread.cpp should fix your segmentation fault.

gmgallo, 2020-08-10:
Isn't that the same compiler deliver with Mbed Studio?

0xc0170, 2020-08-10:
I was referring to some generic C++ online compilers, there are lot of options. Try that to see what runtime error you get, to verify this is not related to the Mbed OS but rather generic issue in the code itself.

gmgallo, 2020-08-10:
You are right, declaring :

extern char my_string[]; 

fixes the issue. I can leave with it, but I think that this behavior is an anomaly not adhering to the C++ standard. Unless I'm missing some fine print left to the compiler implementer choice, a pointer should be resolved as a pointer to the correct object, like it does for any other object. I did try with pointers to several other objects and they all work just fine. When I have the chance I'll run the same test in other platforms and environments.

0xc0170, 2020-08-13:
@gmgallo I'll close this as it's not related to the Mbed OS itself.. report back your findings (it can  be reopened with more details if it is Mbed OS issue).

0xc0170, 2020-08-13:
I actually found an answer that should help: https://stackoverflow.com/questions/6385850/pointer-array-extern-question

closed 2020-08-13 by 0xc0170

gmgallo, 2020-08-13:
I learned something important today. We shouldn't take for granted implicit array to pointer conversions. So, there is nothing wrong with the compiler or Mbed OS. Thanks for taking the time to research it.