ARMmbed/mbed-os #13407: Inconsistent methods used for BufferedSerial and UnbufferedSerial

adbridge, 2020-08-10:
### Description of defect

From the forum:
Why does the new Mbed OS 6.x BufferedSerial use the set_baud and set_format methods when UnbufferedSerial uses the baud and format methods? Since they do the same things with the same arguments, shouldn’t they be named the same?

Looking further it appears that BufferedSerial has it's own direct methods:
```
set_baud(int baud)
set_format(int bits=8, Parity parity=BufferedSerial::None, int stop_bits=1)
```

Whereas UnbufferedSerial has:
```
    using SerialBase::format;
    using SerialBase::attach;
    using SerialBase::baud;
```

where SerialBase has
```
public:
    /** Set the baud rate of the serial port
     *
     *  @param baudrate The baudrate of the serial port (default = 9600).
     */
    void baud(int baudrate);

    /** Set the transmission format used by the serial port
     *
     *  @param bits The number of bits in a word (5-8; default = 8)
     *  @param parity The parity used (SerialBase::None, SerialBase::Odd, SerialBase::Even, SerialBase::Forced1, SerialBase::Forced0; default = SerialBase::None)
     *  @param stop_bits The number of stop bits (1 or 2; default = 1)
     */
    void format(int bits = 8, Parity parity = SerialBase::None, int stop_bits = 1);
```

So this does indeed appear inconsistent.

#### Target(s) affected by this defect ?

All

#### Toolchain(s) (name and version) displaying this defect ?
n/a

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
n/a

#### How is this defect reproduced ? 
n/a

adbridge, 2020-08-10:
@evedon looks like a core issue.

0xc0170, 2020-08-10:
Good find. I did not notice this previously neither. I assume this was intentionally as Bufferedserial is using in case of `format()` SerialBase `format` but adding locks on top of it. We need to look in detail.

cc @ARMmbed/mbed-os-core @kjbracey-arm might also know

LDong-Arm, 2020-08-10:
BufferedSerial and UnbufferedSerial are derived from SerialBase (which can't be instantiated directly). So in my opinion we should
* Align the name `format()` and `baud()` for both buffered and unbuffered serials - in C++ there are no clashes with base class methods of the same names
* Move SerialBase.h into `internal` and make all its methods protected. It's not supposed to be used by applications.

0xc0170, 2020-08-12:
SerialBase is meant to be used, I've seen it in some drivers. Let's dig into the history and find out why the naming is not the same (if there was a specific reason for doing it). @kjbracey-arm is away currently