ARMmbed/mbed-os #13411: SetSysClock fail on STM32L476RGT6

swapnilsayansaha, 2020-08-11:
### Description of defect

++ MbedOS Error Info ++
Error Status: 0x80FF0100 Code: 256 Module: 255
Error Message: Fatal Run-time error
Location: 0x8003AF9
Error Value: 0x0
Current Thread: <unnamed> Id: 0x0 Entry: 0x800042D StackSize: 0x0 StackMem: 0x800045D SP: 0x20017F78 
For more info, visit: https://mbed.com/s/error?error=0x80FF0100&tgt=NUCLEO_L476RG
-- MbedOS Error Info --
SetSysClock failed

#### Target(s) affected by this defect ?

 STM32L476RGT6 clone

#### Toolchain(s) (name and version) displaying this defect ?

Online + Mbed Studio
#### What version of Mbed-os are you using (tag or sha) ?

Mbed OS 2 to 6.

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed studio/online

#### How is this defect reproduced ? 

Just run the blinky sketch on an exact clone of a Nucleo L476RG.

jeromecoutant, 2020-08-11:
Hi
Strange issue... everything seems OK on my side with NUCLEO_L476RG.

LMESTM, 2020-08-11:
> Just run the blinky sketch on an exact clone of a Nucleo L476RG.

What is an exact clone ? 
Can you reproduce the same error on  Nucleo L476RG ?

The error ("SetSysClock failed")  seems to indicate that there is a problem or a difference on the clocks side, like a different LSE or HSE input compared to NUCLEO ?

swapnilsayansaha, 2020-08-11:
> What is an exact clone ?
> Can you reproduce the same error on Nucleo L476RG ?
>
> The error ("SetSysClock failed") seems to indicate that there is a problem or a difference on the clocks side, like a [...]

Hi, 

Yes I debugged the issue and the issue is due to a different LSE model I used for the Nucleo clone. The issue does not occur on the original Nucleo. The original nucleo 6.8pF 32 KHz oscillator while the clone used 4 pF 32 KHz oscillator.

closed 2020-08-11 by swapnilsayansaha

jeromecoutant, 2020-08-12:
As a SW guy, I can't comment, but maybe you can have a look on AN2867 ST Application note:
https://www.st.com/resource/en/application_note/cd00221665-oscillator-design-guide-for-stm8afals-stm32-mcus-and-mpus-stmicroelectronics.pdf

swapnilsayansaha, 2020-08-12:
> As a SW guy, I can't comment, but maybe you can have a look on AN2867 ST Application note:
> https://www.st.com/resource/en/application_note/cd00221665-oscillator-design-guide-for-stm8afals-stm32-mcus-and-mpus-stmicroelectronics.pdf

I actually used that to design the board, but noticed some capacitance issues. In any case I ordered a new PCB with the Nucleo L476RG X2 xtal and caps.

JojoS62, 2020-10-18:
I've seen this issue was closed, but I experienced similiar problems with a H743 board. I assumed also HW or quality problems, but I found it is a missing setting in the software to match the hardware.
The key is in the oscillator design guide doc that @jeromecoutant recommended. The newer STM32 designs have adjustable driver loads for the LSE. This must match the used crystal. On many boards, there are crystals used with higher capacity and higher load capactors. These require more driver power, but the default setting for the MCU is the lowest DL to not damaging the crystal. I will open an issue and it should be decided where to apply this setting. It is hardware dependet and should be adjustable also for custom targets.