ARMmbed/mbed-os #13422: BufferedSerial skips characters in some circumstances

lefebvresam, 2020-08-12:
### Description of defect

When using BufferedSerial **without** NDEBUG=1 mode and with speeds higher than 350000Bd, from time to time serial characters are skipped. When recompiling **with** NDEBUG=1 in mbed_app.json, there is no problem with speeds up to 460800Bd. I tried using priority NVIC_SetPriority(USART3_8_IRQn, 255); but without result.

The parts of the code that invoke this behavior:

The constructor implements a thread and an BufferedSerial object:

```
CommandInterface::CommandInterface(EventQueue* queue, PinName tx, PinName rx, int baud) 
    : _queue(queue), _bufferedserial(tx, rx, baud), _running(false), _response_sent(false),
    _thread(osPriorityNormal, 1024, nullptr, "commandinterface"), _interprete(nullptr) {
        FINE("serial port Tx/Rx pins [%i|%i], Bd [%i], blocking [%i]", tx, rx, baud, _bufferedserial.is_blocking());
        NVIC_SetPriority(USART3_8_IRQn, 255);
    }
```

The thread checks for characters in the software buffer and processes them:

```
// clears a string until n, shrink memory and reset counter
void CommandInterface::clear(std::string& string, size_t n) {
    string.erase(0, n);
    string.reserve(0); // release unnecessary memory to avoid overflows
}

void CommandInterface::serialProcessor() {
    std::string command_buf = "";
    char buf[100] = {0};
    while(_running) {
        uint32_t num = _bufferedserial.read(buf, sizeof(buf));
        printf("\n%i:", num);
        command_buf.append(buf, num);
        printf("%s\n", command_buf.c_str());
        clear(command_buf, string::npos, counter);
    }
    _thread.terminate();
}
```

mbed_app.json:

```
{
    "macros": [
        "NDEBUG=1", --> delete or not delete this line
        "MBED_HEAP_STATS_ENABLED=1",
        "MBED_THREAD_STATS_ENABLED=1",
        "OS_TIMERS=0"
    ],
    "target_overrides": {
        "*": {
            "platform.memory-tracing-enabled": true,
            "platform.callback-nontrivial": true,
            "target.printf_lib": "std"
        },
        "NUCLEO_F091RC": {
            "events.shared-dispatch-from-application": true,
            "rtos.main-thread-stack-size": 5632,
            "drivers.uart-serial-rxbuf-size": 512,
            "drivers.uart-serial-txbuf-size": 256
        }
    }
}
```

A Python script sends a sequence of commands for testing:

```
def pattern_loop(running):
  ...
  while running.is_set():
    if not(q%5): ser.write(b'p1.sil=1\np1.pb0.val=1\np1.sil=0' + closing_seq)
    ...
    elif state == State.IDLE:
    ...  
    time.sleep(fspeed)
```

Serial console output 400000Bd and **not NDEBUG** (skipping chars):

```
33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

32:p1sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

31:p1sil=1
p1.pb0.val=1p1.sil=0ÿÿÿ

32:p1sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

32:p1sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

31:p1sil=1
p1.pb0.val=1
p1.si=0ÿÿÿ
```

Serial console output 350000Bd and **not NDEBUG** (not skipping chars):

```
33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ
```

Serial console output 460800Bd and **NDEBUG=1** (not skipping chars):

```
33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ

33:p1.sil=1
p1.pb0.val=1
p1.sil=0ÿÿÿ
```

#### Target(s) affected by this defect ?

NUCLEO_F091RC

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.2.0
#a2ada74770f043aff3e61e29d164a8e78274fcd4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.4

#### How is this defect reproduced ? 

Create a small project with a similar implementation and check the reason of the issue. You can start with the files attached.

lefebvresam, 2020-08-12:
[commandinterface.zip](https://github.com/ARMmbed/mbed-os/files/5066239/commandinterface.zip)
[verbauwhede.zip](https://github.com/ARMmbed/mbed-os/files/5066243/verbauwhede.zip)

0xc0170, 2020-08-13:
Thanks for the detailed report.

As this was reported with NUCLEO_F091RC, @ARMmbed/team-st-mcd might be interested in this. If the speeds are achievable with debug info enabled.

lefebvresam, 2020-08-13:
The purpose to not use NDEBUG is because in my code I have several MBED_ASSERT lines. With NDEBUG on I can get an unidentified crash because of dangling pointers at some point during testing. However in my opinion the performance may not be affected when not using NDEBUG. I do not know the deeper implementation details but maybe another option is needed with the warning that the performance can be affected for higher serial speeds. A lot of examples work on 460800Bd and I think this should be achievable even in debug mode.

lefebvresam, 2020-08-16:
I also discovered that even when NDEBUG=1 is put in mbed_app.json, speed is 350000Bd and data is bursty coming in, buffered and processed in a separate thread, and at the same time another thread is processing and sending action on the shared main event queue, sometimes characters are dropped also. The result is wrong interpretation of the data, errors and crashes.

```
Settings
---------
Backlight brightness: 39%
Debug port speed: 460800Bd
Command port speed: 350000Bd

Stack statistics (threads) Id         State Priority Size Used
-------------------------- ---------- ----- -------- ---- ----
main                       0x20000C30 2     24       6016 4400
display                    0x200046E0 1     24       512  168
commandinterface           0x20004C18 1     24       1024 264
rtx_idle                   0x20000D80 1     1        512  64
rtx_timer                  0x20000D3C 3     40       768  120
touch                      0x20003658 3     48       1024 200

Heap statistics
---------------
Bytes allocated currently                      : 9398
Max bytes allocated at a given time            : 9422
Cumulative sum of bytes ever allocated         : 9422
Current number of bytes allocated for the heap : 18064
Current number of allocations                  : 50
Number of failed allocations                   : 0

[ERROR   DISPLAY      128] widget name [pb] not found
[ERROR   PAGE         112] page [p1] property [il] not found
[ERROR   DISPLAY      148] property assignment not completed succesfully
```

jeromecoutant, 2021-04-27:
Hi
Could you check if #14532 is improving the situation ?
Thx

lefebvresam, 2021-04-27:
With which branch it can be tested?

wmcelderry, 2021-09-18:
Not sure if this is still open or not?

I have recently reported a similar issue on the mbed community forum which could be related and isn't fixed by #14532 (still happening running mbed-os 6.14.0)

Should I start a new issue or is this enough to link it up?

https://forums.mbed.com/t/uart-losing-data/14577

Thanks for the hard work!

closed 2021-10-20 by ciarmcom