ARMmbed/mbed-os #13424: Mismatch of CY8CKIT-064B0S2-4343W target name

MarceloSalazar, 2020-08-13:
### Description of defect

There is mismatch on the target name between Tools and Mbed OS.
- In mbed-os-tools (mbed-ls) and online database, it's defined as `CY8CKIT064B0S2_4343W`
- In Mbed OS (targets.json), it's defined as `CY8CKIT_064B0S2_4343W` (extra `_`)

This means the target configuration doesn't work out of the box and Mbed CLI is not able to run automated tests.

Proposed fix: update targets.json asap so deliver the change in the upcoming version of Mbed OS.

#### Target(s) affected by this defect ?

CY8CKIT_064B0S2_4343W

#### Toolchain(s) (name and version) displaying this defect ?

Any

#### What version of Mbed-os are you using (tag or sha) ?

master

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

Mbed CLI

#### How is this defect reproduced ? 

```
mbed test -t ARM -m CY8CKIT_064B0S2_4343W --compile
```

This may potentially impact PR https://github.com/ARMmbed/mbed-os/pull/13418

@ARMmbed/team-cypress @ARMmbed/mbed-os-maintainers

vmedcy, 2020-08-13:
See https://github.com/ARMmbed/mbed-os-tools/pull/213#discussion_r373248219 comment from @morser499 :

> I have reduced the length to 20 characters for the new boards.
> Unfortunately this makes the names inconsistent with other boards which can lead to confusion. Hopefully this limit can be addressed sometime soon.

@romanjoe - I wonder how the testing was performed for https://github.com/ARMmbed/mbed-os/pull/13122 - have you happened to use the 'mbedls --mock' command, or manually patched the platform database?

romanjoe, 2020-08-13:
@vmedcy i was using `mbedls -m` approach for testing

jamesbeyond, 2020-08-13:
In mbed OS CI, the test is done in 2 parts, **Build test** (only build greentea binaries) and **Greentea test** (run those binaries on HW).
the mbedls only will be used in Greentea test, but since we don' t have this target in our CI farm, the Greentea test is not run for this targets. Build test passed though.

0xc0170, 2020-08-13:
@romanjoe @vmedcy Can we have this fixed today? 
20 character limit is still present, therefore should be fixed in targets.json

0xc0170, 2020-08-17:
This was not yet fixed, was it ? We created already release candidate for 6.2.1

maclobdell, 2020-08-28:
This issue was highlighted in https://github.com/ARMmbed/mbed-os/issues/13218
@ARMmbed/team-cypress please resolve as soon as possible. This is important to make the board officially supported by Mbed (Mbed Enabled).