ARMmbed/mbed-os #13429: custom targets: target {} has an incomplete target definition

christandlg, 2020-08-14:
### Description of defect

I have created a custom target definition for my project (can be found here: https://bitbucket.org/christandlg/platformio-clicker2-for-stm32 ), but when I try to build for it, mbed throws the error 
```
target Clicker2_for_STM32 has an incomplete target definition
mbed build API internal error
``` 
The target inherits from `FAMILY_STM32`, which is defined in targets\targets.json. I did some poking around inside the mbed framework and I think I have found where this problem comes from.
In .platformio\packages\framework-mbed\tools\targets\__init__.py, function def get_resolution_order(json_data, target_name, order, level=0), the key FAMILY_STM32 cannot be found in the collection of target descriptions. It seems as if the .platformio\packages\framework-mbed\targets\targets.json file is not considered when constructing the target object.

#### Target(s) affected by this defect ?
custom targets

#### Toolchain(s) (name and version) displaying this defect ?
framework-mbed 6.60200.200722 (6.2.0)

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
PlatformIO 4.3.4

#### How is this defect reproduced ?

jeromecoutant, 2020-08-14:
Hi
Yes, since 6.2 version, name has changed!

For your custom_targets.json, I propose:
````
{
    "Clicker2_for_STM32": {
        "inherits": ["MCU_STM32F4"],
        "extra_labels_add": [
            "STM32F407xG"
        ],
        "overrides": {
            "clock_source": "USE_PLL_HSE_XTAL"
        },
        "device_has_add": ["ANALOGOUT", "TRNG"],
        "device_name": "STM32F407VG"
    }
}
````

christandlg, 2020-08-14:
Hi, 
using `MCU_STM32F4` (or your proposed custom_targets.json) did not solve the issue, unfortunately. The error is thrown at the same line in __init__.py, because `MCU_STM32F4` cannot be found. I also noticed that `MCU_STM32F4` is not defined in targets/target.json. 
A user in the PlatformIO forum suggested using `MCU_STM32`, this produced a different error message: ```Failed to extract configuration for CLICKER2_FOR_STM32.
It might not be supported in the this Mbed release.```

jeromecoutant, 2020-08-14:
Oups, you're right
https://github.com/ARMmbed/mbed-os/pull/13229
will be part of next week release...

So for the moment, just change FAMILY_STM32 into MCU_STM32

christandlg, 2020-08-14:
I discussed this issue over [in the PlatformIO forums](https://community.platformio.org/t/mbed-failed-to-extract-configuration-for-custom-board/15413/3), where I found that several steps were necessary to fix this issue:
in platformio.ini:
`"CLICKER2_FOR_STM32": {` (all uppercase characters)
`"inherits": ["MCU_STM32F4"]`
`"default_toolchain": "GCC_ARM",`
removing the src\TARGET_STM32F407xG\device folder and everything contained within as this was just a duplicate of what was already included in mbed and caused compilation issues

closed 2020-08-14 by christandlg