ARMmbed/mbed-os #13437: Error: Use of undeclared identifier 'lwip_gethostbyname'

jakobjackson2016, 2020-08-15:
### Description of defect

#### Target(s) affected by this defect ?
K64F

#### Toolchain(s) (name and version) displaying this defect ?

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 
Import the example: https://os.mbed.com/teams/mbed/code/UDPEchoServer//file/bd7f46b6fa23/ in to your online Mbed Compiler and hit 'Compile'.

0xc0170, 2020-08-17:
The referenced example was deprecated some time ago and it is no longer supported (the last change there was in 2015). Is there a particular reason to use this example and not the latest one using Mbed OS ?

jakobjackson2016, 2020-08-17:
Thanks for your response. The referenced example is currently being used to teach students at our university. I was testing the code and came across this error. Until last year, it used to work in the Mbed Online Compiler but I understand that it is no longer supported. And hence, the error. Do you know by any chance if there's a similar example that will work with the latest Mbed OS? I will also start looking for an alternative example now. Appreciate the clarification.

0xc0170, 2020-08-19:
The online compiler might be a fault here as it uses tools that works across multiple versions and possibly Mbed 2 broke in the recent months (also the online compiler is not being updated anymore, we will update the tools and so on in the upcoming months).

Mbed OS example should work, follow this reference and test it https://os.mbed.com/docs/mbed-os/v6.2/apis/udpsocket.html

0xc0170, 2020-08-20:
I'll close this as won't fix. As stated earlier, the example was deprecated.

Please use the udpsocket from Mbed OS.

closed 2020-08-20 by 0xc0170