ARMmbed/mbed-os #13438: mbed-client-cli: Ridiculously large default buffer size

AGlass0fMilk, 2020-08-16:
### Description of defect

The default size for a line in the `mbed-client-cli` is ridiculously large for any sane command. Even targets with loads of RAM could have a good portion of memory hogged by the cli history.

See below:

https://github.com/ARMmbed/mbed-os/blob/327fe532b2dcec083fdc8abdc36a79070cad7d3a/features/frameworks/mbed-client-cli/source/ns_cmdline.c#L117-L122

2000 bytes? Two THOUSAND!? Per line!? From what I can tell each command is retained in history up to a certain point. The default number of history slots is shown below:

https://github.com/ARMmbed/mbed-os/blob/327fe532b2dcec083fdc8abdc36a79070cad7d3a/features/frameworks/mbed-client-cli/source/ns_cmdline.c#L129-L134

So after running 32 commands the `mbed-client-cli` framework is perpetually hogging 64kB of RAM -- enough to choke an nRF52832 completely.

I recently discovered this while trying to run BLE and a command line on an nRF52832_DK. Only one command would work, all following commands would just print nothing...

When I debugged I found that the command line library was failing to dynamically allocate a new 2000-byte buffer for the new command...

Furthermore, the `mbed_lib.json` for this library has no configuration options. This likely isn't a priority due to the upcoming tools updates.

#### Target(s) affected by this defect ?
All

#### Toolchain(s) (name and version) displaying this defect ?
GCC ARM

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
Try using `mbed-client-cli` library on any target with less than a metric tonne of RAM.

loverdeg, 2020-08-16:
Explains a lot...

0xc0170, 2020-08-17:
@artokin  would you be able to help reviewing ?

artokin, 2020-08-17:
Master repository for mbed-client-cli is https://github.com/ARMmbed/mbed-client-cli and some configuration updates are already available there. Unfortunately those changes are not back ported to Mbed OS yet.

As a workaround, would it be possible to use `macros` section in application configuration file to overwrite those problematic constants?

artokin, 2020-08-17:
PR https://github.com/ARMmbed/mbed-os/pull/13443 created to update Mbed OS with the latest mbed-client-cli

teetak01, 2020-08-18:
The `use_minimum_set `feature in the mbed-client-cli is specifically for constrained device use, which is part of the #13443

0xc0170, 2020-09-03:
The fix was merged to Mbed Os. I'll close this as resolved.

closed 2020-09-03 by 0xc0170