ARMmbed/mbed-os #13440: update to psoc6pdl-1.6.1 broke `CYTFM_064B0S2_4343W`

moshe-shahar, 2020-08-16:
### Description of defect

After merging https://github.com/ARMmbed/mbed-os/pull/13386, `CYTFM_064B0S2_4343W` system init failed with [hard fault](https://github.com/ARMmbed/mbed-os/blob/327fe532b2dcec083fdc8abdc36a79070cad7d3a/targets/TARGET_Cypress/TARGET_PSOC6/psoc6pdl/drivers/source/cy_pra.c#L1829)

#### Target(s) affected by this defect ?
`CYTFM_064B0S2_4343W`

#### Toolchain(s) (name and version) displaying this defect ?
`GCC_ARM`

#### What version of Mbed-os are you using (tag or sha) ?

[327fe53](https://github.com/ARMmbed/mbed-os/tree/327fe532b2dcec083fdc8abdc36a79070cad7d3a)

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ? 

1. Clone https://github.com/ARMmbed/mbed-os-example-psa
1. Apply the following to mbed_app.json
```
diff --git a/mbed_app.json b/mbed_app.json
index c8be573277..1903278312 100644
--- a/mbed_app.json
+++ b/mbed_app.json
@@ -12,6 +12,11 @@
         },
         "CY8CKIT_064S2_4343W": {
             "rtos.thread-user-stack-size": 16384
+        },
+        "CYTFM_064B0S2_4343W": {
+            "rtos.thread-user-stack-size"               : 16384,
+            "platform.stdio-baud-rate"                  : 115200,
+            "target.policy_file"                        :"policy_multi_CM0_CM4_jitp.json"
         }
     },
     "macros": [
```
3. Build for `CYTFM_064B0S2_4343W`

0xc0170, 2020-08-16:
@ARMmbed/team-cypress Please review

0xc0170, 2020-08-16:
I would like to understand how this affects the planned release (code freeze 2 days ago). If this is a blocker, we need to have this fixed.

cc @andypowers @ARMmbed/mbed-os-maintainers

0xc0170, 2020-08-16:
@ARMmbed/team-cypress I would like to find out how many targets are affected - the scope of this issue. Are more targets affected?

The recent nightly do not contain greentea tests ( see http://mbed-os-ci.s3-website-eu-west-1.amazonaws.com/?prefix=jenkins-ci/ARMmbed/mbed-os/mbed-os-ci-nightly/artifacts/master/604/ ) - I already reported this to the test team.

vmedcy, 2020-08-17:
Hello, I guess the hard fault is expected as part of PRA driver design:

https://github.com/ARMmbed/mbed-os/blob/e95c64f/targets/TARGET_Cypress/TARGET_PSOC6/psoc6pdl/drivers/source/cy_pra.c#L150

psoc6pdl-1.6.0 provides PRA 1.0, psoc6pdl-1.6.1 - PRA 1.1. The version of PRA driver on CM0+ and Cm4 should be identical.

This information is available in PRA driver changelog: https://cypresssemiconductorco.github.io/psoc6pdl/pdl_api_reference_manual/html/group__group__pra.html

All PSoC 64 targets affected.

CC @avptin @romanjoe  @ifyall

0xc0170, 2020-08-17:
@vmedcy Thanks for the details, regarding

> psoc6pdl-1.6.0 provides PRA 1.0, psoc6pdl-1.6.1 - PRA 1.1. The version of PRA driver on CM0+ and Cm4 should be identical.

does that mean we got 2 different PRA or how shall I read it? How can this be fixed?

vmedcy, 2020-08-17:
> How can this be fixed?

Update CM0+ images to use PRA 2.0 as well. I don't know where these images are located and what is the update procedure. @romanjoe , @avptin - do you know what needs to happen to keep CM0+ part of PRA solution in sync?

romanjoe, 2020-08-17:
@vmedcy i need to generate new new psoc6cm0p asset package, compiled with new version of psoc6pdl-1.6.1, then we need to test it and then i can make a PR to mbed-os with that

moshe-shahar, 2020-08-17:
AFAIK, @CharleyChu is working on a fix for `CYTFM_064B0S2_4343W`.
I tested with his tfm_s.hex and it fixes this issue.

0xc0170, 2020-08-17:
> AFAIK, @CharleyChu is working on a fix for CYTFM_064B0S2_4343W.
I tested with his tfm_s.hex and it fixes this issue.

💯  great, will one target only be fixed ? I see in our tree 2 Cypress targets containing tfm_s.hex file

moshe-shahar, 2020-08-17:
I don't know what is the status of `CY8CKIT_064S2_4343W` target.

0xc0170, 2020-08-17:
If this could be fixed today (a pull request created - we will run through the CI), we can generate rc2 otherwise if nothing else pops up, we would stay with rc1.

0xc0170, 2020-08-17:
@CharleyChu @RaymondNgun shall we expect a patch today ? If not, we will proceed with rc1.

RaymondNgun, 2020-08-17:
@0xc0170, yes... that is the plan for CYTFM_064B0S2_4343W.  Thanks!

PR #13447 (2020-08-18): psoc64: Update TF-M release image
Upgrade PDL to v1.6.1 to resolve #13440

### Summary of changes 

#### Impact of changes 

#### Migration actions required
commit ace81a8b (2020-08-18): psoc64: Update TF-M release image

Upgrade PDL to v1.6.1 to resolve #13440
closed 2020-08-18 by 0xc0170