ARMmbed/mbed-os #13458: Mbed OS cellular example not working for custom MC60 module

kirtiraje, 2020-08-18:
### Description of defect

I am using Mbed OS version 6.2 on Mbed CLI. I have used below example
https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-cellular/

#### What version of Mbed-os are you using (tag or sha) ?

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

#### How is this defect reproduced ?

0xc0170, 2020-08-19:
I don't have an experience with cellular but the first question when it faults - Would you be able to trace back where it faults and see what triggers the fault? To get to context , it might be invalid data or state.

kirtiraje, 2020-08-19:
As per the logs it is not even started to communicate to the MC60 module. if the module is powered ON then it will send the response to AT command. I am not sure if my configuration setting for interfacing to MC60 in mbed-app.json file are correct.
Also I did not find the POWER KEY pin mentioned in the driver code for powering ON the module.  POWER PIN need to be keep low for around  1 sec to power up/power down the module.

logs are shown below:
Built: Aug 19 2020, 12:53:48

[MAIN], plmn: NULL
Establishing connection

[2K[39m[1ms][INFO][CELL]: New CellularContext  (0x1FFF1808)[0m

[2K[39m[1ms][INFO][CELL]: CellularContext plmn NULL[0m

[2K[39m[2ms][INFO][CELL]: CellularCont
++ MbedOS Fault Handler ++

FaultType: HardFault

kirtiraje, 2020-08-20:
Hi, I have tested the same code in online compiler and the MC60 module is powered up. But it is not getting connected to the network.
AT+CSQ is 99,99.

adbridge, 2020-08-24:
@kirtiraje  "https://os.mbed.com/teams/mbed-os-examples/code/mbed-os-example-cellular/" This is not a version of mbed-os...

kirtiraje, 2020-09-02:
Hi, The Mbed OS cellular example is working on the FRDM K64 dev board with the online compiler. It is still not working with Mbed CLI environment. I will check the Mbed Cli version ,update it and will again test it.

closed 2020-09-02 by kirtiraje