ARMmbed/mbed-os #13480: PwmOut resets settings after suspend.

lefebvresam, 2020-08-24:
### Description of defect

When using PwmOut and putting the settings to 4kHz/25% it starts to output the right PWM-signal, but as soon as you do a suspend and resume, the frequency will be reset to 50Hz and you need to redo the settings again. Maybe it's a suggestion to use a start/stop functionality which is separate from the configuration settings.

#### Target(s) affected by this defect ?

NUCLEO_G474RE

#### Toolchain(s) (name and version) displaying this defect ?

GCC_ARM

#### What version of Mbed-os are you using (tag or sha) ?

mbed-os-6.2.0
#a2ada74770f043aff3e61e29d164a8e78274fcd4

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)

mbed-cli 1.10.4

#### How is this defect reproduced ? 

Create a small project to test:

```
PwmOut buzzer(PC_0);

void stopBuzzer() {
    buzzer.suspend();
}

void startBuzzer() {
    buzzer.resume();
}

void initBuzzer() {
    buzzer.period_us(250);
    buzzer.write(0.25f);
    buzzer.suspend();
}

int main()
{
    Timeout t;
    initBuzzer();

    ThisThread::sleep_for(5s);

    startBuzzer();
    t.attach(&stopBuzzer, 2s);

    while(true) ThisThread::sleep_for(1s);
}
```

LMESTM, 2020-08-24:
From a quick look at pwmout.cpp and #11641 it seems that only duty cycle is saved and restored, not the period / frequency. I guess also the period needs to be saved and restored.

lefebvresam, 2020-08-24:
Indeed. I see on the scope that my duty cycle stays on 25% but the frequency changes from 4kHz to 50Hz.

jeromecoutant, 2020-08-24:
@lefebvresam Do you think you could propose some update and complete #11641 ?

@hugueskamba @kjbracey-arm

lefebvresam, 2020-08-25:
@jeromecoutant I'm not actively working on the mbed project.

jeromecoutant, 2020-08-26:
@lefebvresam could you check #13492

closed 2020-09-10 by 0xc0170