ARMmbed/mbed-os #13482: QSPIFBlockDevice fails to read SFDP-Header after updating to mbed-os 6.2.0

DBS06, 2020-08-24:
### Description of defect

After updating from mbed-os 5.15.1 to mbed-os-6.2.0 the `QSPIFBlockDevice` fails at `QSPIFBlockDevice::init()` with an error code of `-4001`. 

To reproduce this behavior I tried out the blockdevice-example (github.com/ARMmbed/mbed-os-example-blockdevice) and run it on the NRF52840_DK board and it fails too. With mbed-os-5.15.1 the initialization runs through and works as expected. To verify if it has something to do with the new `QSPIFBlockDevice-Driver` and `SFDP` implementation, I hard replaced the `QSPIFBlockDevice` with the version from mbed-os-5.15.1 and the blockdevice-example works also as expected.

Therefore I looked inside the new libs and I could enclose the issue to `sfdp_parse_headers()` from `SFDP.h`, but unfortunately I get no time to dig further into this issue and it is really hard to debug this stuff, because the `-Og` flag optimizes out a lot of variables and error handling goto-jumps. I tried to compile only with `-O0` but this fails somehow at startup.

Furthermore I think I found a copy-and-paste-error inside `
mbed-os/components/storage/blockdevice/COMPONENT_QSPIF/mbed_lib.json` the default for 

```
        "MCU_NRF52840": {
            "QSPI_FREQ": "32000000",
            "QSPI_MIN_READ_SIZE": "4",
            "QSPI_MIN_PROG_SIZE": "256"
        }, 
```
should be
```
        "MCU_NRF52840": {
            "QSPI_FREQ": "32000000",
            "QSPI_MIN_READ_SIZE": "4",
            "QSPI_MIN_PROG_SIZE": "4"
        }, 
```

#### Target(s) affected by this defect ?
NRF52840

#### Toolchain(s) (name and version) displaying this defect ?
gcc-arm-none-eabi-9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
Try out the blockdevice-example (github.com/ARMmbed/mbed-os-example-blockdevice) on the NRF52840_DK  once with mbed-os 5.15.1 and once with 6.2.0.

loverdeg, 2020-08-24:
@ARMmbed/team-embeddedplanet

DBS06, 2020-10-15:
We solved it by using the "old" QSPIF-BD.

closed 2020-10-15 by DBS06