ARMmbed/mbed-os #13483: minimal-printf does not print leading zeros

DBS06, 2020-08-24:
### Description of defect
The output of:
`printf("val: 0x%02X\n, 0x0A);`
with minimal-printf is:
`val: 0xA`
but it should be:
`val: 0x0A`

with setting `"target.printf_lib": "std"` the output is as expected.

#### Target(s) affected by this defect ?
NRF52840

#### Toolchain(s) (name and version) displaying this defect ?
gcc-arm-none-eabi-9-2019-q4-major

#### What version of Mbed-os are you using (tag or sha) ?
mbed-os-6.2.0

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
`printf("val: 0x%02X\n, 0x0A);`

JojoS62, 2020-08-27:
This is documentated behaviour, but the same applies to the missing precision modifier. This is already a FAQ in the Mbed forum.
It is very inconvenient to modify a string with this formatting afterwards and this feature should not add too much complexity in the minimal-printf.

DBS06, 2020-09-03:
documentated behaviour

closed 2020-09-03 by DBS06