ARMmbed/mbed-os #13498: NUCLEO_G031K8 Failed to build all greentea images in bare-metal profile

jamesbeyond, 2020-08-26:
### Description of defect

after  NUCLEO_G031K8 been added to mbed-os, we found that  NUCLEO_G031K8 failed on building greentea `TESTS-MBED_HAL-MINIMUM_REQUIREMENTS`
#### Target(s) affected by this defect ?
 NUCLEO_G031K8

#### Toolchain(s) (name and version) displaying this defect ?
GCC_ARM only, ARMC6 works fine

#### What version of Mbed-os are you using (tag or sha) ?

latest master : afcf91f3314d82ce8d555f4a891c9fd8db13910f

#### What version(s) of tools are you using. List all that apply (E.g. mbed-cli)
mbed-cli

#### How is this defect reproduced ? 
run:
```
mbed test --compile -m NUCLEO_G031K8 -t GCC_ARM --app-config TESTS/configs/baremetal.json
```
Error:
```
Build failures:
  * NUCLEO_G031K8::GCC_ARM::TESTS-MBED_HAL-MINIMUM_REQUIREMENTS
        Building project minimum_requirements (NUCLEO_G031K8, GCC_ARM)
        Scan: GCC_ARM
        Scan: minimum_requirements
        Compile [100.0%]: main.cpp
        Link: minimum_requirements
        /usr/local/gcc-arm-none-eabi-9-2019-q4-major/bin/../lib/gcc/arm-none-eabi/9.2.1/../../../../arm-none-eabi/bin/ld:./BUILD/tests/NUCLEO_G031K8/GCC_ARM/./TESTS/mbed_hal/minimum_requirements/.link_script.ld:90 cannot move location counter backwards (from 00000000200010b8 to 0000000020001000)
        collect2: error: ld returned 1 exit status
        
[mbed] Working path "/builds/ws/mbed-os-ci_build-GCC_ARM@3/mbed-os" (program)
[mbed] WARNING: Could not find mbed program in current path "/builds/ws/mbed-os-ci_build-GCC_ARM@3/mbed-os".
       You can fix this by calling "mbed new ." in the root of your program.
---
[mbed] ERROR: "/usr/bin/python" returned error.
       Code: 1
       Path: "/builds/ws/mbed-os-ci_build-GCC_ARM@3/mbed-os"
       Command: "/usr/bin/python -u /builds/ws/mbed-os-ci_build-GCC_ARM@3/mbed-os/tools/test.py -D MBED_TRAP_ERRORS_ENABLED=1 -D SKIP_TIME_DRIFT_TESTS=1 -D MBED_TEST_MODE -t GCC_ARM -m NUCLEO_G031K8 --source . --build ./BUILD/tests/NUCLEO_G031K8/GCC_ARM --test-spec ./BUILD/tests/NUCLEO_G031K8/GCC_ARM/test_spec.json --build-data BUILD/tests/build_data.json --app-config TESTS/configs/baremetal.json --greentea"
       Tip: You could retry the last command with "-v" flag for verbose output
```

0xc0170, 2020-08-26:
cc @AGlass0fMilk @ARMmbed/team-st-mcd

jeromecoutant, 2020-08-26:
Strange...
Building echo test is ok...

jeromecoutant, 2020-08-26:
Adding in json:
        "overrides": { "boot-stack-size": "0x400" },
makes build ok

Why there is a difference between GCC and ARMC6 ?

0xc0170, 2020-08-31:
As it is blocking now #13452, how to unblock it ? It is important fix to targets. 

Is target that tiny it does not even build hal minimal req test?

jeromecoutant, 2020-09-01:
> As it is blocking now #13452, how to unblock it ? 

As I said, solution is to set "boot-stack-size" to 0x400"
But, question was , why behavior is different for ARM and GCC ?

Thx

jamesbeyond, 2020-09-03:
> > As it is blocking now #13452, how to unblock it ?
> 
> As I said, solution is to set "boot-stack-size" to 0x400"
> But, question was , why behavior is different for ARM and GCC ?
> 
> Thx

I think the default value for `boot-stack-size` is 0x1000 for baremetal , see 
https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L60-L62
and
https://github.com/ARMmbed/mbed-os/blob/master/targets/TARGET_STM/TARGET_STM32G0/TARGET_STM32G031xx/TOOLCHAIN_ARM/stm32g031xx.sct#L29-L30

for RTOS it just got override to 0x400:
https://github.com/ARMmbed/mbed-os/blob/master/cmsis/device/rtos/mbed_lib.json#L65

Maybe ARMC6 not failed just because the code size compile by ARMC6 slightly smaller than GCC ?

On other size constrained targets, I saw people forced override that value in target.json
https://github.com/ARMmbed/mbed-os/blob/master/targets/targets.json#L253-L255
maybe we should do the same to this target ?

PR #13587 (2020-09-10): MCU_STM32G031xx : decrease boot-stack-size
### Summary of changes 

This fixes #13498 

As a small RAM target, default boot-stack-size is decreased for baremetal full support.

#### Impact of changes 

#### Migration actions required 

### Documentation 

----------------------------------------------------------------------------------------------------------------
### Pull request type
closed 2020-09-10 by 0xc0170